tpaviot/pythonocc-core #302: Tests Fail

erniejunior, 2016-09-12:
Hi,
I just tried to build pythonOCC on debian. Everything works fine but then the tests and all examples fail with the same error:
```
Traceback (most recent call last):
  File "run_tests.py", line 22, in <module>
    import core_wrapper_features_unittest
  File "/home/m/tmp/pythonocc-core/test/core_wrapper_features_unittest.py", line 24, in <module>
    from OCC.Standard import Standard_Transient, Handle_Standard_Transient
  File "/usr/lib/python2.7/dist-packages/OCC/Standard.py", line 32, in <module>
    _Standard = swig_import_helper()
  File "/usr/lib/python2.7/dist-packages/OCC/Standard.py", line 28, in swig_import_helper
    _mod = imp.load_module('_Standard', fp, pathname, description)
ImportError: dynamic module does not define init function (init_Standard)
```
I built OCE 0.16.1 myself (no errors during the build) and am running swig 2.0.12. The pythonOCC version I used was 0.16.5 (4f4c324).
Any ideas on how to resolve the issue would be highly appreciated.

tpaviot, 2016-09-13:
There might be a conflict between python versions. You try to run pythonocc with py2.7, are you sure you compiled it for this version of python ?

erniejunior, 2016-09-13:
I do not know for what version I compiled it (I did not change anything in the CMakeLists file) but  `PYTHONOCC_INSTALL_DIRECTORY` is set to `/usr/lib/python2.7/dist-packages/OCC` and running the tests with python3 did not help either.
But thanks a lot so far for this hint!

jf---, 2016-09-13:
@erniejunior , did you build pythonocc yourself or have you installed using conda?

erniejunior, 2016-09-13:
I built it myself.

tpaviot, 2016-09-14:
@erniejunior Ok for PYTHON_INSTALL_DIRECTORY. Could you please report the values of PYTHON_LIBRARY and PYTHON_INCLUDE_DIR please ?

erniejunior, 2016-09-14:
I could not find the `PYTHON_INSTALL_DIRECTORY` right away but the following lines were in the `CMakeCache.txt` in the build direcotry:
```
PYTHON_EXECUTABLE:FILEPATH=/usr/bin/python
PYTHON_INCLUDE_DIR:PATH=/usr/include/python3.4m
PYTHON_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libpython3.4m.so
```
This looks wrong because /usr/bin/python points to python 2.7.9 but the other paths seem to point to python 3.4 (which I have also installed)

tpaviot, 2016-09-14:
@erniejunior ok, that's what I thought: it's a conflict between python versions. You linked pythonocc to python3.4 library, and run it using python2. This must be consistent. I think the default python (/usr/bin/python) is python3.4. If you want to compile pythonocc for python2, specify PYTHON_EXECUTABLE, PYTHON_INCLUDE and PYTHON_LIBRARY. You can use cmake-gui, it's much practical to work with the cmake GUI.

erniejunior, 2016-09-14:
Thanks a lot for this hint!
On my debian system the default python points to python2 (not 3 like in most modern distributions). I will now try to guess sensible values for the variables mentioned above. Maybe you can give me a hint? Then I will recompile.

tpaviot, 2016-09-14:
On your system, the following should work:
```
$ mkdir cmake-build
$ cd cmake-build
$ cmake -DPYTHON_EXECUTABLE:/usr/bin/python -DPYTHON_INCLUDE:/usr/include/python2.7 -DPYTHON_LIBRARY:/usr/lib/x86_64-linux-gnu/libpython2.7.so ..
```
Not completely sure about the inc and lib paths though.

erniejunior, 2016-09-14:
I am currently running a build (80%) with guessed paths for python3. I will try your version too as soon as the current build finishes.

tpaviot, 2016-09-22:
@erniejunior Any update for this issue ?

tpaviot, 2016-09-23:
Closing the issue, it seems you've been doing something wrong.

closed 2016-09-23 by tpaviot