tpaviot/pythonocc-core #303: Module GEOMAlgo

jirihnidek, 2016-09-12:
Hi,
was module `GEOMAlgo` removed or was it renamed? This blog post mentions `GEOMAlgo_Splitter`:

http://www.pythonocc.org/ecosystem/use-cases/pythonocc-and-smesh/

I would like to use it in my application. I tried to google it, but I didn't find any useful information.
Thanks.

jf---, 2016-09-12:
hi @jirihnidek ,

Correct, the GEOM module ( from Salome ) isnt maintained currently.
Splitting shapes however is perfectly possible though, see the examples.
Let me know if you have a specific use case.

tpaviot, 2016-09-13:
@jirihnidek I removed the outdated post from the blog. The GEOM addition to OCE/OCCT is not maintained anymore.

closed 2016-09-13 by tpaviot

jirihnidek, 2016-11-09:
hi @jf--- 

sorry for late response. I have specific use case. I can split one solid shape into the two solid shapes and glue them together using sewing:

[image]

but when I do meshing the the shape in GMSH, then meshing is compatible at lines, but it is not compatible at bordering face and it is problem

[image]

Thanks for advice.

jf---, 2016-11-09:
its appears that when meshing the shape is not considered a TopoDS_Solid or TopoDS_Compound ,but rather 2 separate shells ( which don't require vertices to be shared on shared edges )

so I'm curious if the resulting file contains what kind of topology.
that would help answer the question whether GMSH can create a mesh with shared edges from compound or solid.

trelau, 2017-04-01:
@jf--- @jirihnidek I implemented Salome 8.2.0 GEOMAlgo_Splitter and submitted a PR [here](https://github.com/tpaviot/pythonocc-core/pull/411).

jirihnidek, 2017-04-04:
Hi, I solved the problem using boolean operations and reshaping. You can look at my progress in this issue : https://github.com/GeoMop/bapprox/issues/10#issuecomment-288761796. The source code can be found here: https://github.com/GeoMop/PythonOCC_Examples/blob/master/src/bspline_solid_boolean.py ... I'm not longer member of this project.

Cheers.

jf---, 2017-04-05:
@jirihnidek ,  a PR on that example would be a welcome contrib! 
a very interesting project...

tpaviot, 2017-04-06:
@jirihnidek I was not aware of the GeomMp project or the PythonOCC_Examples. There are many very interesting things, feel free to submit your examples so that they become part of pythonocc.

jirihnidek, 2017-04-06:
Hi, @tpaviot . I'm no longer member of GeoMop project. I work at Red Hat since 1st April. I'm poking @janakolajaehlerova for submitting these examples :-). She needs to practice using git and github :-).

jf---, 2017-04-10:
share @tpaviot feeling wrt to seeing many inspired examples, something of a gold mine ;)