tpaviot/pythonocc-core #307: Bundle SMESH with pythonOCC

Puffton, 2016-09-22:
Would it be possible to bundle SMESH with pythonOCC in Conda?

Currently I get the following with today's release of pythonOCC:
```python
>>> from OCC.SMESH import SMESH_Gen
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named SMESH
```

tpaviot, 2016-09-23:
@smesh The SMESH fork is not maintained anymore by its original author. Maintaining, packaging, wrapping smesh is a huge task. I wish I had time to do it myself, but it must be done by someone else if we ever want smesh to be back into pythonocc. I don't plan any officiel smesh support within pythonocc, I'll work on it according to my free time. If you wish to contribute to the pythonocc project, managing the smesh issue is a good way to go.

e-dub, 2016-09-23:
This is related to the issue that I started #289, albeit it is without the aspect of bundling for Conda.

@tpaviot: What do you think the expected workload of getting the current state of the SMESH wrappers to function in this version of pythonOCC?
@Puffton: Is something you would be interested in cooperating on?

Puffton, 2016-09-23:
I have implemented a solution using numpy-stl today. In time though, it would be great if we could use SMESH with pythonOCC (I have code for that as well, which I haven't tested yet). I'm going to work on other projects in a couple of weeks, but the person taking over after me may be interested in contributing to this project. Since it's not a pressing issue for us right now, it would be a while (months at least) before then.

tpaviot, 2017-06-06:
smesh support available in master from PR #432

closed 2017-06-06 by tpaviot