tpaviot/pythonocc-core #308: Changed behaviour of GeomAPI_ExtremaCurveSurface in pythonocc 0.17

kngbuzzo, 2016-09-23:
Hi everyone,
I am using pythonocc to compute the intersection points between a B-spline surface  (Geom_BSplineSurface object) and a set of vectors (Geom_Line objects), using Geom_ExtremaCurvedSurface. This used to work like a charm in version 0.16.5, but after installing version 0.17.0 several intersections fail:
[image]
The number of failures seems to be somehow dependent on the length of the line evaluation interval [0,wmax]. Also, the surface's circumferential coordinate starts and ends at the top where the failures occur, so I don't know if this might have something to do with it. GeomAPI_IntCS manages to find the points, but is much slower than Geom_ExtremaCurvedSurface.

Have there been any changes in the function between versions which could explain this?

tpaviot, 2016-09-23:
@kngbuzzo Thank you for this feedback. pythonocc-0.17 is based upon oce-0.17.2, which contains many changes since the previous oce-0.16 (FYI this is the step from official OCCT671 to OCCT680). OCCT680 release notes (https://www.opencascade.com/sites/default/files/documents/Release_Notes_6.8.0.pdf) does not mention any change to the  Geom_ExtremaCurvedSurface function/algorithm, however there might have been many other changes to functions/classes used by  the Geom_ExtremaCurvedSurface class. Your use case seems to be complex, would it be possible that you provide a simple example that reproduces the issue ?

tpaviot, 2016-09-23:
@kngbuzzo Are you sure about the spelling of Geom_ExtremaCurvedSurface, I can't even fin such a class in pyocc-0.16 or 0.17

kngbuzzo, 2016-09-23:
@tpaviot Thanks for your answer. Geom_ExtremaCurvedSurface was a typo and should say Geom_ExtremaCurveSurface as in the title. Sorry about that. I have tried a second example, which is just a regular cylindrical barrel, and all the intersections work fine.

tpaviot, 2016-09-23:
@kngbuzzo It will be *very* difficult to debug without an example that show the error. There might have some parameters to tune, but unfortunately I don't use to work with this class and can't be helpfull.

kngbuzzo, 2016-09-23:
@tpaviot I'm sorry I don't have a good example to share. The algorithm seems to work great with all my simpler models... Btw. I went through the release notes and they mention new global optimization algorithms for extrema computation in their highlights. More specifically, the Extrema_GenExtCS class is said to have been changed, which, I suspect, is somehow being used by GeomAPI_ExtremaCurveSurface. In a way this makes sense, because the more I think about it I am certain that my problem is related to the initial guess for the optimization, which is what has been changed, from my understanding. So I guess that answers at least part of my original question.

jf---, 2016-10-29:
@kngbuzzo , we appreciate your feedback, but lets have an example, or close this issue; its not in an actionable state... 

without being able to reproduce the issue, this will take too much resources on our end I'm afraid...

kngbuzzo, 2016-11-02:
@jf--- thanks for bringing this back up and sorry for the delay. I came up with an example to illustrate my problem based on an ellipsoid, which you can find here: https://github.com/kngbuzzo/pythonocc_examples For 2500 intersections I get the following output:
pythonocc 0.16.5:
```
Time for intersections 0.480000019073
[]
0 failures
```
[image]

pythonocc 0.17:
```
Time for intersections 2.21000003815
[[  2.00000000e-01   3.00321910e-04   0.00000000e+00]
 [  1.15714286e+01   3.79488064e-01   4.89295676e-02]
    ...
 [  5.21836735e+01   4.00839802e-01  -5.16825693e-02]
 [  2.00000000e-01   3.00321910e-04  -7.35576530e-20]]
107 failures
```
[image]

The black dots indicate failures (obviously). I hope this is a little more useful to you.

jf---, 2016-11-02:
that's perfect @kngbuzzo , thanks so much! 
Can you share the script please, than I can delve deeper into this, whether this can be overcome in pythonocc or should be escalated to OpenCascade. 

Perhaps you can share it as at gist.github.com, or just paste it here if you like...

Thanks a lot for this!

kngbuzzo, 2016-11-02:
@jf--- , I added the script in a repository under my name, the link to which I have provided above (<https://github.com/kngbuzzo/pythonocc_examples>). Is that ok for you?

jf---, 2016-11-02:
Shoot, I misread your earlier comment, sorry @kngbuzzo & thanks again!!

kngbuzzo, 2016-11-03:
Cool, thanks for loking into it @jf--- !

rainman110, 2017-02-01:
@kngbuzzo  I think this issue should be posted directly at the [OCCT Mantis](https://tracker.dev.opencascade.org/my_view_page.php)? Did you already do it?

We could provide an even more simplified example
 - one, where the computation time is much slower than before
 - one, where the intersection is failing

kngbuzzo, 2017-02-13:
@rainman110 No, I haven't posted anything yet. Will they be able to work with the example in Python?

Following your suggestions, I have added two more test cases to the example script.

rainman110, 2017-02-13:
@kngbuzzo We should create a minimalistic  C++ example for the OCCT guys. I could provide help, if you want. Great with the new test cases!

kngbuzzo, 2017-02-14:
@rainman110 If you could give me a hand with that, that'd be great.

tpaviot, 2020-01-05:
@kngbuzzo What's up with this issue. Did you try your code with more recent pythonocc release ?

kngbuzzo, 2020-02-05:
@tpaviot I ran the updated version of the code using versions 0.18.2 and 7.4.0-rc1 to these results:

[image]
[image]

OCC 7 is clearly improving the situation (possibly at the cost of performance), but the fundamental issue remains.

jf---, 2020-02-17:
@jnwalther is it feasible that the failing samples ( impressive plot btw ) are an [Umbilical point](https://en.wikipedia.org/wiki/Umbilical_point)?

jnwalther, 2020-05-12:
@jf--- I went ahead and checked my example surface for umbilical points. The source data is an ellipsoid with a circular cross section (i.e. b==c), which means it should only have an umbilical point at the tip. I compared the principal curvature distibutions for an analytical ellipsoid to my test surface, which brought to light some problems wrt the continuity. After fixing this, I get the plot below, which suggests that the B-Spline surface accurately represents the location of the umbilical points (Coloring is W as definied by [Maekawa et al.](https://doi.org/10.1016/0167-8396(95)00018-6); the criterion for an umbilical point is W==0).

[image]

Fixing the surface also brought down the number of failures further to
- OCC6: 72
- OCC7: 13

However, I have to report, that while the failures occur more often close the umbilical points, they do not occur there exclusively, as shown in the figure below.
[image]

rainman110, 2020-05-12:
@jnwalther I forgot to follow up this. Let’s simply post this issue at the official occt bug tracker. PythonOCC can’t fix this issue unfortunately.
Could you extract the failing points, or even better, write a python test script comparing with the expected values? I’ll translate this then into c++

jnwalther, 2020-05-12:
@rainman110 I agree we should take this issue upsteam. If you could help me out with the C++ part, that'd be great. As requested, I put together a minimal version of the test here: https://github.com/jnwalther/pythonocc_examples/blob/master/test_minimal.py
Let me know if you can work with this or if you need anything else.

rainman110, 2020-05-12:
@jnwalther Could you add to your script also the expected values? This could lead to an even much better test checking not only failure, but also accuracy.

jnwalther, 2020-05-12:
@rainman110 By expected values you mean uvw?

rainman110, 2020-05-12:
Either this or the xyz coordinates

jnwalther, 2020-05-12:
Ah, but the xyz coordinates are already there: https://github.com/jnwalther/pythonocc_examples/blob/master/expected_points.csv
The file is referred to in the script.

rainman110, 2020-05-12:
Ah okay... I thought, the epected point could be derived from the ellipse equation, but this list is probably also okay.

jnwalther, 2020-05-12:
Ah ok, I understand. Maybe using the list not so bad though. Otherwise we'd have to consider how well the example surface approximates an actual ellipsoid.

tpaviot, 2024-05-23:
Old issue, please reopen if still valid with latest release

closed 2024-05-23 by tpaviot