tpaviot/pythonocc-core #309: how to make class "BRepFeat_MakeCylindricalHole" work correctly

chentao807, 2016-09-23:
Hi, i'm trying to make a hole feature with class BRepFeat_MakeCylindricalHole

```python
def add_feature(base):
    d = 0.8
    origin = gp_Ax1(gp_Pnt(0, 0, 0), gp_Dir(0, 0, 1))
    hole = BRepFeat_MakeCylindricalHole()
    hole.Init(base, origin)
    hole.Perform(d/2)
    return hole.Shape()
```

but 'BRepFeat_MakeCylindricalHole' object has no attribute 'Shape'
please tell me, how to make class work correctly

tpaviot, 2016-09-26:
@chentao807 Please provide a complete example that I can run on my machine, it will save my time rather than having to finish the code above

jf---, 2016-09-26:
here's a full example:

```python
from OCC.BRepFeat import BRepFeat_MakeCylindricalHole
from OCC.BRepPrimAPI import BRepPrimAPI_MakeSphere
from OCC.Display.SimpleGui import init_display
from OCC.gp import gp_Ax1, gp_Pnt, gp_Dir

display, start_display, add_menu, add_function_to_menu = init_display()

sph = BRepPrimAPI_MakeSphere(10).Shape()

def add_feature(base):
    d = 0.8
    origin = gp_Ax1(gp_Pnt(0, 0, 0), gp_Dir(0, 0, 1))
    hole = BRepFeat_MakeCylindricalHole()
    hole.Init(base, origin)
    # hole.Init(base, origin)
    hole.Perform(d/2)
    # hole.Images()
    # < Swig Object of type 'BOPCol_DataMapOfShapeListOfShape *' at 0x1114f82d0 >
    # hole.Splits()
    # <Swig Object of type 'BOPCol_DataMapOfShapeListOfShape *' at 0x10d31f150>

add_feature(sph)

start_display()

```

Looks like overlap with #162 , where the `.Images` method returns a `BOPCol_DataMapOfShapeListOfShape` which is not wrapped

jf---, 2016-09-26:
@tpaviot , would the new __review/BOPCol-fix__ branch fix this? Is it time to test this?

chentao807, 2016-09-26:
idea come from
http://www.algotopia.com/contents/opencascade/opencascade_basic

```python
from OCC.gp import gp_Pnt, gp_Dir, gp_Ax2, gp_Ax1
from OCC.BRepPrimAPI import BRepPrimAPI_MakeSphere
from OCC.BRepFeat import BRepFeat_MakeCylindricalHole
from OCC.Display.SimpleGui import init_display
import math

def generate_shape():
    radius = 1.0
    a = math.atan(0.5)
    origin = gp_Ax2(gp_Pnt(0, 0, 0), gp_Dir(0, 0, 1))

    builder = BRepPrimAPI_MakeSphere(origin, radius, -a, a)
    return builder.Shape()

def add_feature(base):
    d = 0.8
    origin = gp_Ax1(gp_Pnt(0, 0, 0), gp_Dir(0, 0, 1))
    hole = BRepFeat_MakeCylindricalHole()
    hole.Init(base, origin)
    hole.Perform(d/2)

    # BRepFeat_MakeCylindricalHole' object has no attribute 'Shape'
    return hole.Shape()

if __name__ == '__main__':
    disp, loop, m, a = init_display()
    sphere = generate_shape()

    shape = add_feature(sphere)
    disp.DisplayShape(shape, update=True)
    disp.FitAll()
    loop()
```

tpaviot, 2016-09-27:
@jf--- Maybe. Actually I found out that some missing methods comre from the fact that the BOPCol wrapper is empty (i.e. no class is wrapped). Thus all classes that inherit from a BOPCol_* miss a lot of useful methods. Wrapping BOPCol is not that easy, that's certainly why I did that in a recent past.

tpaviot, 2016-09-27:
@chentao807 Ok thank you I'll check that. When it is fixed, I will add this snippet to our example suite.

tpaviot, 2016-09-27:
@jf--- The branch is not ready for test yet. I will push a PR when ready

jf---, 2016-09-27:
@tpaviot , why not [port the cool example of algotopia?](http://www.algotopia.com/contents/opencascade/opencascade_sprocket)

it would be a cool non-trivial example

tpaviot, 2016-10-11:
@chentao807  @jf---  BRepFeat_MakeCylindricalHole' object has no attribute 'Shape' because the wrapper lacks the BOPAlgo_BuilderShape class (and a few other related). Indeed, the Shape() method is inherited from this class. They're not wrapped because these classes have some protected destructor defined (which confuses SWIG). Branch https://github.com/tpaviot/pythonocc-core/tree/review/BOPCOl-BOPAlgo-fix is a fix attempt.

jf---, 2016-10-11:
@tpaviot , is the mentioned branch ready for review / for general review / specific for this BRepFeat_MakeCylindricalHole example? Just so to know what to expect...

closed 2016-10-13 by tpaviot