tpaviot/pythonocc-core #312: cannot supply Graphic3d_RenderingParams argument

jf---, 2016-09-26:
this is while working on a raytrace view example:

```python
from __future__ import print_function

from OCC.BRepPrimAPI import BRepPrimAPI_MakeSphere
from OCC.Display.SimpleGui import init_display
from OCC.Graphic3d import Graphic3d_RenderingParams

display, start_display, add_menu, add_function_to_menu = init_display()

ais_boxshp = None

def build_shape():
    boxshp = BRepPrimAPI_MakeSphere(50.0).Shape()
    ais_boxshp = display.DisplayShape(boxshp, update=True)
    return ais_boxshp

build_shape()

params = display.View.RenderingParams()

render_params = Graphic3d_RenderingParams()
display.View.ChangeRenderingParams(render_params)

start_display()

#    display.View.ChangeRenderingParams(render_params)
# TypeError: V3d_View_ChangeRenderingParams() takes exactly one argument (2 given)
```

tpaviot, 2016-09-27:
@jf--- it has something to do with byref parameters in the Graphic3d.i SWIG file

jf---, 2016-09-27:
OK, cool, lets do #314 first?

tpaviot, 2017-09-18:
Fixed by commit 53f2496ef71ce0ddb1a061c0bf811311cb0fb077

closed 2017-09-18 by tpaviot