tpaviot/pythonocc-core #317: gp_QuaternionSLerp doesnt take arguments

jf---, 2016-09-26:
```python
from OCC.gp import gp_Vec, gp_Quaternion, gp_QuaternionSLerp

vX = gp_Vec(12, 0, 0)
vY = gp_Vec(0, 12, 0)
v45 = (gp_Vec(1, 1, 1).Normalized() * 12)

q = gp_Quaternion()
q1 = gp_Quaternion(vX, vX)
q2 = gp_Quaternion(vX, vY)

interp = gp_QuaternionSLerp(q1, q2)

for i in range(100):
    i__ = i / 100.
    interp.Interpolate(i__, q)
    print("i: ", i__)
    print("quat X,Y,Z,W: {} / {} / {} / {}".format(q.X(), q.Y(), q.Z(), q.W()))

```

---

quat X,Y,Z,W: 0.0 / 0.0 / 0.0 / 1.0
i:  0.96
quat X,Y,Z,W: 0.0 / 0.0 / 0.0 / 1.0
i:  0.97
quat X,Y,Z,W: 0.0 / 0.0 / 0.0 / 1.0
i:  0.98
quat X,Y,Z,W: 0.0 / 0.0 / 0.0 / 1.0
i:  0.99
quat X,Y,Z,W: 0.0 / 0.0 / 0.0 / 1.0

looks like the by-ref argument `q` is not updated?

tpaviot, 2016-09-26:
The python wrapper is incorrect:
```
void Interpolate (Standard_Real theT,gp_Quaternion  theResultQ);
```
should be
```
void Interpolate (Standard_Real theT,gp_Quaternion  &theResultQ);
```

tpaviot, 2016-09-26:
Branch review/Quat-fix should fix the problem.

closed 2016-09-27 by jf---

tpaviot, 2016-09-28:
Fixed by PR #322

PR #322 (2016-09-28): gp by reference fixes
This branch adds back by ref passing/return in the gp module. gp is now considered as other modules.

Fixes #317