tpaviot/pythonocc-core #321: Font_BRepFont support

jf---, 2016-09-26:
would it be possible to wrap the [Font_BRepFont](https://www.opencascade.com/doc/occt-6.9.0/refman/html/class_font___b_rep_font.html) class? This would allow fonts / text to be used as a BRep primitive... 

I've worked on a freetype based python solution, but realized that there's a proper class in OCE

tpaviot, 2016-10-11:
@jf---  Can you please test the branch https://github.com/tpaviot/pythonocc-core/tree/review/Font-wrapper  and report your experiments/screenshots.

This branch has a major drawback: it adds freetype as a dependency. This dependency has to be removed before a possible merge into the master branch.

jf---, 2016-10-11:
@tpaviot wow, that was swift! I'll get back to you tomorrow morning, thanks again for the fast response. Wrt freetype, isnt this a required dependency of OCE?

tpaviot, 2016-10-11:
@jf--- it is

jf---, 2016-10-12:
hi Thomas,

So the build passes on linux, but not on osx alas, see the [failing osx build on that branch here](https://travis-ci.org/tpaviot/pythonocc-core/jobs/166875222). I ran into similar problems:

```bash
[ 30%] Building CXX object CMakeFiles/_FairCurve.dir/Darwin/x86_64-MinSizeRel-64/FairCurvePYTHON_wrap.cxx.o
Scanning dependencies of target _Font
[ 30%] Building CXX object CMakeFiles/_Font.dir/Darwin/x86_64-MinSizeRel-64/FontPYTHON_wrap.cxx.o
Linking CXX shared module Darwin/x86_64-MinSizeRel-64/_FairCurve.so
ld: warning: path '/Users/jelleferinga/miniconda/envs/_build/lib/libfreetype.dylib' following -L not a directory
[ 30%] Built target _FairCurve
[ 31%] Swig source
/Users/jelleferinga/miniconda/conda-bld/work/Darwin/x86_64-MinSizeRel-64/FontPYTHON_wrap.cxx:7025:7: error: 
      'operator delete' is a protected member of 'Standard_Transient'
      delete arg1;
      ^
/Users/jelleferinga/miniconda/envs/_build/OCE.framework/Versions/0.17/Resources/../../../../include/oce/Font_BRepFont.hxx:41:23: note: 
      constrained by protected inheritance here
class Font_BRepFont : protected Font_FTFont
                      ^~~~~~~~~~~~~~~~~~~~~
/Users/jelleferinga/miniconda/envs/_build/OCE.framework/Versions/0.17/Resources/../../../../include/oce/Standard_Transient_proto.hxx:36:5: note: 
      member is declared here
    DEFINE_STANDARD_ALLOC
    ^
/Users/jelleferinga/miniconda/envs/_build/OCE.framework/Versions/0.17/Resources/../../../../include/oce/Standard_DefineAlloc.hxx:61:9: note: 
      expanded from macro 'DEFINE_STANDARD_ALLOC'
  void  operator delete (void* theAddress)                             \
        ^
1 error generated.
make[2]: *** [CMakeFiles/_Font.dir/Darwin/x86_64-MinSizeRel-64/FontPYTHON_wrap.cxx.o] Error 1
make[1]: *** [CMakeFiles/_Font.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

this overlaps with the Travis build...

Shoot, I know its tricky to debug on platforms your not running...

tpaviot, 2016-10-13:
I tried a fix, removing wrapper for Font_BRepFont destructor

tpaviot, 2016-10-13:
@jf--- this seems to work, you can go on with testing

jf---, 2016-10-13:
hi @tpaviot , cool, yes, there is a `OCC.Font` module now, see:

[image]

but alas:

[image]

We're getting closer!

tpaviot, 2016-10-27:
@jf--- Font_BRepFont is quite a strange clas. It inherits from the Font_FTFont, with a protected constructor that confuses SWIG. Rather than wrapping the entire Font package, another solution is to add C++ code to the Visualization module, related to font display.

tpaviot, 2016-10-27:
@jf--- This is definitely the base way to proceed see review/font-brep-rendering

I added an example:
[image]

jf---, 2016-10-27:
Thomas! Wow, that's awesome!
I'm really thrilled to see this come about!

I have just one comment though; I would consider Text a primitive, like BRepPrimAPI_Box / Cylinder / Sphere etc... so the `Visualization` seems a strange place actually...

I guess that  the `Visualization` module is original to pythonocc and therefore practical to use...

But what about the following; sometimes being able to dive down to C++ and expose a class is actually the right way to go ( or; route of least resistance ). So having such a corner where we can expose bespoke C++ functionality specifically for pythonocc is actually a really good idea.

so perhaps `from OCC.Addons import BRepText` ?

So thrilled to see this!

tpaviot, 2016-10-27:
@jf--- you're right. So far, Visualization is just a convenient container for this experiment. OCC.Addons is ok, I'll do this

jf---, 2016-10-27:
Thanks @tpaviot! Will build as soon as you get there ;)
Great to see this come about, really cool!

tpaviot, 2016-10-27:
@jf--- I refactored the branch so that a Addons module is available. The Visualization module is left as is, the Font code is in the Addons. As a consequence, I also moved up the Display package, at the same level as Addons and Visualization. See https://github.com/tpaviot/pythonocc-core/tree/review/font-brep-rendering

closed 2016-10-31 by jf---