tpaviot/pythonocc-core #324: TBB related issue building current master on osx

jf---, 2016-09-27:
@tpaviot , congrats on the recent refactor of the CI / conda scripts, its great to be able to use these locally, really neat...

that said, while building I've ran into the following; it seems that intel TBB is linked.
this is peculiar:

- it seems like there are no flags to set / remove linking to TBB
- TBB is not part of the dependencies in `meta.yaml`
- FYI I'm using DLR-SC's OCE 0.17.2 build, which utilizes TBB
- see the img at the bottom, wrt how TBB's path

```python
+ python run_tests.py
Traceback (most recent call last):
  File "run_tests.py", line 22, in <module>
    import core_wrapper_features_unittest
  File "/Users/jelleferinga/miniconda/conda-bld/work/test/core_wrapper_features_unittest.py", line 25, in <module>
    from OCC.Standard import Standard_Transient, Handle_Standard_Transient
  File "/Users/jelleferinga/miniconda/envs/_test/lib/python3.5/site-packages/OCC/Standard.py", line 22, in <module>
    _Standard = swig_import_helper()
  File "/Users/jelleferinga/miniconda/envs/_test/lib/python3.5/site-packages/OCC/Standard.py", line 21, in swig_import_helper
    return importlib.import_module(mname)
  File "/Users/jelleferinga/miniconda/envs/_test/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: dlopen(/Users/jelleferinga/miniconda/envs/_test/lib/python3.5/site-packages/OCC/_Standard.so, 2): Library not loaded: libtbbmalloc.dylib
  Referenced from: /Users/jelleferinga/miniconda/envs/_test/lib/python3.5/site-packages/OCC/_Standard.so
  Reason: image not found
TESTS FAILED: pythonocc-core-0.17-py35_5
```

[image]

[image]

tpaviot, 2016-09-28:
@jf--- Intel TBB is not used directly by pythonocc, no pythonocc library should be linked to TBB. Maybe this has something to do with @rainman110 compilation for osx64.

rainman110, 2016-09-28:
@jf--- @tpaviot I am linking oce against tbb. During installation of OCE, OCE writes the file lib/oce-0.17/OCE-libraries-release.cmake, where all linked libraries are listed. Whenever you link against OCE with cmake, your project (pythonocc) is automatically linked against the dependencies of OCE.

Therefore, you get an explicit link to tbb in ```_BRep.so```.

@jf--- In your case it seems, that the rpath to tbb is not correctly set in ```_BRep.so```. In the build.sh file of the conda recipe, we are fixing some rpaths for python and freetype. This has to be done as well for tbb.

What I find strange is, that these problems did not occur during CI and also not on my mac locally. There, the rpaths are correctly set and tbb is found.

As you can see, TKVoxel links correctly to tbb, as ```@rpath/./``` is prefixed before the tbb libraries. In ```_BRep.so``` this is not the case. I am not sure, why this happens on your computer, but not in the CI and on mine.

rainman110, 2016-09-28:
By the way, I think we can fix all these issues and the binary relocation problematics by properly installing the swig generated *.so files. Currently, the CMake script just copies them to the target destination:

https://github.com/tpaviot/pythonocc-core/blob/master/CMakeLists.txt#L284

When performing an actual installation (using ```TARGET``` instead of ```FILES```), CMake does some magic to make things correct.

It did that also for our tigl app. Have a look to the changes of the CMakeLists.txt:
https://github.com/DLR-SC/tigl-conda/blob/master/tigl/fixrpath.patch

I think, we can do the same with pythonocc-core. Then we don't need to fix up the rpaths manually anymore.

tpaviot, 2016-09-28:
@rainman110 It would be great

jf---, 2016-09-28:
@rainman110 , yes, so [macdependecy](https://github.com/kwin/macdependency/releases) was created as a mac replacement for dependecywalker, its a nice gui on top of otool...

tpaviot, 2016-10-04:
This should be fixed by #331 @rainman110 can you confirm ?

jf---, 2016-10-11:
@tpaviot confirmed

closed 2016-10-11 by jf---