tpaviot/pythonocc-core #325: port algodia examples

jf---, 2016-09-27:
[see here](http://www.algotopia.com/contents/opencascade/opencascade_sprocket)

chentao807, 2016-09-27:
as you wish, @jf--- 
https://gist.github.com/chentao807/b1774deed8eacf009c8492947d283a04

Just work, but no best.  Make anything better if you can

jf---, 2016-09-27:
nice work @chentao807.... thanks 💐 

so I could copy the snippet, but probably best is if you make a pull request, this way you'll show up as a contributor to the project... 

we wouldnt want to steal the hard earned karma ;)

awesome, code looks mint if you ask me 🏆

jf---, 2016-09-27:
Awesome @chentao807 

[image]

tpaviot, 2016-09-28:
This is quite an amount of work, well done @chentao807 Please open a PR so that this is included in our example suite. 2 modifications reqsuested:

* add a license header (see the files in the /examples directory) with your name/email

* we use underscores to name functions. You should move all your functions names to underscores (buildTooth->build_tooth etc.)

rainman110, 2016-09-28:
@chentao807 Wow, great work! :tada:  I also modelled gears ones and I know how much work this can be :smirk:

tpaviot, 2016-10-04:
There is one more example to port see http://www.algotopia.com/contents/opencascade/opencascade_basic

jf---, 2016-10-04:
@tpaviot OK on it...

closed 2016-10-13 by tpaviot