tpaviot/pythonocc-core #326: Transparency Z order bug?

lalvarezt, 2016-09-27:
Hi

[image]

When I represent two transparent objects, one them is always in front of the other, it doesn't matter the angle of view. In the picture the red cylinder is in front of the green shape, but it gets represented on the back.

Also I'm getting the ragged edges on some shapes, even with antialising on.

Is this a pythonocc bug or OCE? Or is there a way to fix it?

Thanks in advance

jf---, 2016-09-27:
good question, it would be easier to delve into this with an example that reproduces the problem. 
btw, the screenshot is looking good besides the transparency issue. can i ask how you managed to render the edges on top of the shaded geometry?

jf---, 2016-09-27:
@lalvarezt , perhaps [this offers a clue](https://www.opencascade.com/doc/occt-6.9.0/refman/html/class_v3d___view.html): see **EnableDepthTest**

jf---, 2016-09-27:
can reproduce with:

```python
def cut(event=None):
    # Create Box
    Box = BRepPrimAPI_MakeBox(200, 60, 60).Shape()
    # Create Sphere
    Sphere = BRepPrimAPI_MakeSphere(gp_Pnt(100, 20, 20), 80).Shape()
    # Cut: the shere is cut 'by' the box
    Cut = BRepAlgoAPI_Cut(Sphere, Box).Shape()
    display.EraseAll()
    ais_box = display.DisplayShape(Box)
    display.Context.SetTransparency(ais_box, 0.8)
    ais_cut = display.DisplayColoredShape(Cut, color="CYAN")
    display.Context
    display.Context.SetTransparency(ais_cut, 0.4)
    display.FitAll()
```

strange thing is that `display.View.IsDepthTestEnabled() -> True`

[image]

lalvarezt, 2016-09-27:
@jf--- Thanks for the reply, to render the edges on to of the shaded geometry use
`self.OCE_widget.context.DefaultDrawer().GetObject().SetFaceBoundaryDraw(True)`

lalvarezt, 2016-09-27:
@jf--- So I guess, we'll have to wait for the new GLSL / raytracing. Thanks

jf---, 2016-09-27:
@lalvarezt , it looks like it, though I agree that it looks shockingly aweful...

`self.OCE_widget.context.DefaultDrawer().GetObject().SetFaceBoundaryDraw(True)` 🍹 many thanks, an excellent find...

since we're on the topic, your screenshot ( yes, I read your post ) looks pretty good to me... have you done more trick wrt lighting & shading?

tpaviot, 2017-05-04:
@lalvarezt  cant' reproduce the issue, either running oce-0.17 or 0.18. I added an example in commit 686ed9ac943f39f2f95d3ff6063473abdca5688b, can't see anything wrong. Can you please provide a test case

tpaviot, 2017-05-21:
@lalvarezt ping

closed 2020-01-05 by tpaviot