tpaviot/pythonocc-core #337: move examples to their corresponding directory

jf---, 2016-10-04:
@tpaviot we've spoken about this before, I think it would be cool to have all `core_geometry_` examples in a `geometry` folder, which has a `README.md`. The advantage here is that we could have a folder of screenshots in the `pythonocc-core/doc/examples/_images/` folder. 

This way, when browsing Github, you're presented with a great looking gallery of what pythonocc is capable of.

Polish, true, but shiny is pretty 🍹

tpaviot, 2016-10-12:
There's no need to move examples to different directories to generate screenshots and/or a readme.md The examples/screenshots and/or the readme.md should be automatically generated from the examples. 

Have a look at mrdoob threejs examples folder https://github.com/mrdoob/three.js/tree/dev/examples as well as the https://threejs.org/examples/ page. This is pretty good.

jf---, 2016-10-12:
hi @tpaviot ,

wrt generated, that's obvious the right approach.
that said, we could think of a html gallery, with such an outline on the left, a code viewer next to it, with a webgl view next to that... i have a server that could run things

jf---, 2016-10-12:
ok, perhaps its something to look into...

closed 2017-06-10 by tpaviot