tpaviot/pythonocc-core #338: BoundingBox dimension

maciekgroch, 2016-10-06:
Hey,
I am using PythonOCC to modify my design. As a matter of fact I also experimented with [PyGeM](http://www.pythonocc.org/ecosystem/related_projects/free-form-deformation-with-pygem/) in order to apply free form deformation. 

I tried to implement a [Bounding Box](https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_geometry_utils.py) so that the whole object gets modified -- I tried to emulate the BoundingBox modificiation in Rhino (hope you know it).

Unofrtunately, I encountered following problem: created bounding box has inappropriate dimensions. When a bounding box is created, the dimensions (min(x, y, z) and max(x, y, z)) are calculated from the control points of an object (this is what I understand). However, not all control points are part of a solid or closed polysurface (when a solid is created faces are trimmed).

 Is there a way to create a BB that would take into account the dimensions of a solid, not all the control points??

Thank you in advance.

jf---, 2016-10-06:
hi @maciekgroch ,

So I could point you in the direction of how to make such a BoundingBox, such that it can be deformed with a gizmo. But lets talk boundingbox first. The problem is clear; would a minimal boundingbox ( AABB or oriented bounding box ) solve your problem?

jf---, 2016-10-06:
minimal bounding box
---

import sys
from OCCUtils.edge import Edge
from OCCUtils.Common import vertex2pnt
from OCCUtils.Topology import Topo
from OCC.gp import gp_Vec
import numpy as np

def oriented_boundingbox(shape, sampling_distance=50.0):
    """
    :param shape: TopoDS_Shape

    :param sampling_distance: edges are discretized, this is the distance by which the curve is divided in segments

    :return: four gp_Vec, describing an oriented bounding topods_shape [ aka minimal bounding topods_shape ]
    the first argument is the lowest corner point, than follow the X, Y & Z vectors that describe the
    axises of the oriented bounding topods_shape
    """
    from scipy import spatial

    vertices = []
    for e in Topo(shape).edges():
        edg = Edge(e)
        a, b = map(lambda x: vertex2pnt(x, as_tuple=True), [edg.first_vertex(), edg.last_vertex()])
        vertices.append(a)
        vertices.append(b)

        zzz = [i[1] for i in
               edg.divide_by_number_of_points(edg.length() / sampling_distance, as_tuple=True)]

        for vertex in zzz:
            # no duplications
            if not vertex in vertices:
                vertices.append(vertex)

    vertices = np.array(vertices)
    delaunay_mesh = spatial.Delaunay(vertices)

    shift = delaunay_mesh.points.min(axis=0)
    points = delaunay_mesh.points - shift

    min_volume = sys.maxint

    for i in delaunay_mesh.convex_hull.tolist():

        tri = delaunay_mesh.points[i]
        e1 = tri[0] - tri[1]
        e2 = tri[0] - tri[2]
        e3 = tri[1] - tri[2]
        _edges1 = [e1, e2, e3]
        _edges2 = [e2, e3, e1]

        for a, b in zip(_edges1, _edges2):

            # transform the two edges into a orthogonal basis

            w = vec_cross(a, b)  # returns normalized vector
            u = vec_cross(w, a)
            v = vec_cross(u, w)

            # project all the points on to the basis u1, u2 u3
            # p = calcProjections(points, u1, u2, u3)
            p = calcProjections(points, [u, v, w])

            volume, mins, maxes = calcVolume(p)

            # we are looking for the minimum volume topods_shape
            if volume <= min_volume:
                min_volume = volume
                specs = u, v, w, mins, maxes, volume

    u, v, w, mins, maxes, volume = specs

    # get the corner by using our projections, then shift it to move
    # it back into the same origin as the original set of points
    corner = u * mins[0] + v * mins[1] + w * mins[2]
    corner += shift

    # create the sides which are vectors with the magnitude the length
    # of that side
    v1 = u * (maxes[0] - mins[0])
    v2 = v * (maxes[1] - mins[1])
    v3 = w * (maxes[2] - mins[2])
    return map(lambda x: gp_Vec(*x), [corner, v1, v2, v3])

def calcVolume(p):
    """Calculates the volume of the topods_shape that would encompass the given
    points using the given projection. projection is sized (NxM) where
    N is the number of points and M is the number of vectors they were
    projected onto. Also return the minimum and maximum bounds along
    each of those vectors."""

    # the minimum and maximum projection of each basis vector
    mins = p.min(axis=0)
    maxes = p.max(axis=0)

    # the volume product of each difference between the maximum and
    # minimum values from the projection onto each basis vector
    volume = np.prod(maxes - mins)

    return volume, mins, maxes

def calcProjections(points, vectors):
    """Calculates the projection of points (NxD) onto the vectors
    (MxD) and return the projections p which is a matrix sized (N, M)
    where N is the number of points and M is the number of vectors.
    p[i][j], is the projection of points[i] onto vectors[j] (which is
    between 0 and 1)."""

    u = np.array(vectors)

    # project the points onto the vectors into on fell swoop
    d = np.dot(points, u.T)

    # this is the dot product of each vector with itself
    v2 = np.diag(np.inner(u, u))

    p = d / v2

    return p

def vec_cross(u, v):
    """Return the normalized cross product of u and v."""
    w = np.cross(u, v)
    w /= np.sqrt(np.sum(v ** 2))
    return w

---

def make_oriented_box(v_corner, v_x, v_y, v_z):
    """
    produces an oriented box
    oriented meaning here that the x,y,z axis do not have to be
    cartesian aligned

    :param v_corner: the lower corner
    :param v_x: gp_Vec that describes the X-axis
    :param v_y: gp_Vec that describes the Y-axis
    :param v_z: gp_Vec that describes the Z-axis
    :return: TopoDS_Solid
    """
    from OCC.BRepOffsetAPI import BRepOffsetAPI_MakePipe
    verts = map(lambda x: x.as_pnt(), [v_corner, v_corner+v_x, v_corner+v_x+v_y, v_corner+v_y])
    p = make_polygon(verts, closed=True)
    li = make_line(v_corner.as_pnt(), (v_corner+v_z).as_pnt())
    ooo = BRepOffsetAPI_MakePipe(p, li)
    ooo.Build()
    shp = ooo.Shape()

    bottom = make_face(p)
    top = translate_topods_from_vector(bottom, v_z, True)
    oriented_bbox = make_solid(sew_shapes([bottom, shp, top]))
    return oriented_bbox

    corner, v1, v2, v3 = oriented_boundingbox(sh)
    box = make_oriented_box(corner, v1, v2, v3)
    display.DisplayColoredShape(box, "WHITE")

maciekgroch, 2016-10-06:
I tried to run the methods `oriented_boundingbox`, however there is a problem: 
`TypeError: vertex2pnt() got an unexpected keyword argument 'as_tuple'`, so I simply deleted this argument... well i didn't solve the issue. Are you sure the code is right? I have up-to-date master branch pulled.

jf---, 2016-10-06:
hi @maciekgroch , so the latest pyocc version does deal with the .Coord method a little different. But that's the very latest master. so yes, its possible some calls might require minor tweaking.

maciekgroch, 2016-10-10:
Hey @jf--- ! I implemented the example you proposed. Unfortunately, it is a _universal_ solution. IMHO it is because the BB dimensions are based on positions of edges of the shape. In one of the test cases I used a shape made of a hemisphere and other elements. The BB matched only the edges, the hemisphere was left untouched.

jf---, 2016-10-10:
ah, interesting!
so it would be useful to have a default sampling algorithm ( the current? ) and be able to supply custom one?

maciekgroch, 2016-10-10:
It is mostly based on the code you provide. Unfortunately, due to the company policy I mustn't provide the test shapes I used for testing...

```

import sys

import numpy as np
from OCCUtils.edge import Edge
from OCCUtils.Construct import *
from OCCUtils.Topology import Topo
from OCC.gp import gp_Vec

def oriented_boundingbox(shape, sampling_distance=50.0):
    """
    :param shape: TopoDS_Shape

    :param sampling_distance: edges are discretized, this is the distance by which the curve is divided in segments

    :return: four gp_Vec, describing an oriented bounding topods_shape [ aka minimal bounding topods_shape ]
    the first argument is the lowest corner point, than follow the X, Y & Z vectors that describe the
    axises of the oriented bounding topods_shape
    """
    from scipy import spatial

    vertices = []
    for e in Topo(shape).edges():
        edg = Edge(e)
        a, b = map(lambda x: vertex2pnt(x), [edg.first_vertex(), edg.last_vertex()])
        vertices.append(a)
        vertices.append(b)

        zzz = [i[1] for i in
               edg.divide_by_number_of_points(edg.length() / sampling_distance)]

        for vertex in zzz:
            # no duplications
            if not vertex in vertices:
                vertices.append(vertex)

    vertices = map(lambda x: (x.X(), x.Y(), x.Z()), vertices)
    delaunay_mesh = spatial.Delaunay(vertices)

    shift = delaunay_mesh.points.min(axis=0)
    points = delaunay_mesh.points - shift

    min_volume = sys.maxint

    for i in delaunay_mesh.convex_hull.tolist():

        tri = delaunay_mesh.points[i]
        e1 = tri[0] - tri[1]
        e2 = tri[0] - tri[2]
        e3 = tri[1] - tri[2]
        _edges1 = [e1, e2, e3]
        _edges2 = [e2, e3, e1]

        for a, b in zip(_edges1, _edges2):

            # transform the two edges into a orthogonal basis

            w = vec_cross(a, b)  # returns normalized vector
            u = vec_cross(w, a)
            v = vec_cross(u, w)

            # project all the points on to the basis u1, u2 u3
            # p = calcProjections(points, u1, u2, u3)
            p = calcProjections(points, [u, v, w])

            volume, mins, maxes = calcVolume(p)

            # we are looking for the minimum volume topods_shape
            if volume <= min_volume:
                min_volume = volume
                specs = u, v, w, mins, maxes, volume

    u, v, w, mins, maxes, volume = specs

    # get the corner by using our projections, then shift it to move
    # it back into the same origin as the original set of points
    corner = u * mins[0] + v * mins[1] + w * mins[2]
    corner += shift

    # create the sides which are vectors with the magnitude the length
    # of that side
    v1 = u * (maxes[0] - mins[0])
    v2 = v * (maxes[1] - mins[1])
    v3 = w * (maxes[2] - mins[2])
    return map(lambda x: gp_Vec(*x), [corner, v1, v2, v3])

def calcVolume(p):
    """Calculates the volume of the topods_shape that would encompass the given
    points using the given projection. projection is sized (NxM) where
    N is the number of points and M is the number of vectors they were
    projected onto. Also return the minimum and maximum bounds along
    each of those vectors."""

    # the minimum and maximum projection of each basis vector
    mins = p.min(axis=0)
    maxes = p.max(axis=0)

    # the volume product of each difference between the maximum and
    # minimum values from the projection onto each basis vector
    volume = np.prod(maxes - mins)

    return volume, mins, maxes

def calcProjections(points, vectors):
    """Calculates the projection of points (NxD) onto the vectors
    (MxD) and return the projections p which is a matrix sized (N, M)
    where N is the number of points and M is the number of vectors.
    p[i][j], is the projection of points[i] onto vectorsj."""

    u = np.array(vectors)

    # project the points onto the vectors into on fell swoop
    d = np.dot(points, u.T)

    # this is the dot product of each vector with itself
    v2 = np.diag(np.inner(u, u))

    p = d / v2

    return p

def vec_cross(u, v):
    """Return the normalized cross product of u and v."""
    w = np.cross(u, v)
    w /= np.sqrt(np.sum(v ** 2))
    return w

def make_oriented_box(v_corner, v_x, v_y, v_z):
    """
    produces an oriented box
    oriented meaning here that the x,y,z axis do not have to be
    cartesian aligned

    :param v_corner: the lower corner
    :param v_x: gp_Vec that describes the X-axis
    :param v_y: gp_Vec that describes the Y-axis
    :param v_z: gp_Vec that describes the Z-axis
    :return: TopoDS_Solid
    """
    from OCC.BRepOffsetAPI import BRepOffsetAPI_MakePipe
    verts = map(lambda x: x.as_pnt(), [v_corner, v_corner+v_x, v_corner+v_x+v_y, v_corner+v_y])
    p = make_polygon(verts, closed=True)
    li = make_line(v_corner.as_pnt(), (v_corner+v_z).as_pnt())
    ooo = BRepOffsetAPI_MakePipe(p, li)
    ooo.Build()
    shp = ooo.Shape()

    bottom = make_face(p)
    top = translate_topods_from_vector(bottom, v_z, True)
    oriented_bbox = make_solid(sew_shapes([bottom, shp, top]))
    return oriented_bbox

```

jf---, 2016-10-10:
hi @maciekgroch , thanks, would you be so kind as to indicate changes?

today I ran into the following which might be of interest:

```python
def brepbndlib_Add(*args):
    """
    * Adds the shape S to the bounding box B. More precisely are successively added to B: - each face of S; the triangulation of the face is used if it exists, - then each edge of S which does not belong to a face, the polygon of the edge is used if it exists - and last each vertex of S which does not belong to an edge. After each elementary operation, the bounding box B is enlarged by the tolerance value of the relative sub-shape. When working with the triangulation of a face this value of enlargement is the sum of the triangulation deflection and the face tolerance. When working with the polygon of an edge this value of enlargement is the sum of the polygon deflection and the edge tolerance. Warning - This algorithm is time consuming if triangulation has not been inserted inside the data structure of the shape S. - The resulting bounding box may be somewhat larger than the object.

    :param S:
    :type S: TopoDS_Shape &
    :param B:
    :type B: Bnd_Box &
    :param useTriangulation: default value is Standard_True
    :type useTriangulation: bool
    :rtype: void

    """
    return _BRepBndLib.brepbndlib_Add(*args)
```

possible using the triangulation argument is interesting?

maciekgroch, 2016-10-11:
Hi @jf---, to be honest the only changes I made regarded `vertex2pnt(x, as_tuple=True)` -> `vertex2pnt(x)` and then creating tuples on my own:  `vertices = map(lambda x: (x.X(), x.Y(), x.Z()), vertices)` . I guess these changes are the most imporant. 

I tried the example you provided with` brepbndlib_Add` method. Unfortunately it doesn't work the way expected either. Frankly speaking I developed something easy based on the mesh of a shape.

maciekgroch, 2016-10-11:
Actually what can be done is the following: ```

from OCC.Bnd import Bnd_Box
from OCC.BRepBndLib import brepbndlib_Add
from OCC.gp import gp_Vec
from OCCUtils.Topology import Topo

def get_boundingbox(shape, tol=1e-6, as_vec=False):
    """ return the bounding box of the TopoDS_Shape `shape`
    Parameters
    ----------
    shape : TopoDS_Shape or a subclass such as TopoDS_Face
        the shape to compute the bounding box from
    tol: float
        tolerance of the computed boundingbox
    as_vec : bool
        wether to return the lower and upper point of the bounding box as gp_Vec instances
    Returns
    -------
        if `as_vec` is True, return a tuple of gp_Vec instances
         for the lower and another for the upper X,Y,Z values representing the bounding box
        if `as_vec` is False, return a tuple of lower and then upper X,Y,Z values
         representing the bounding box
    """
    bbox = Bnd_Box()
    bbox.SetGap(tol)
    topo = Topo(shape)
    for face in list(topo.faces()):
        brepbndlib_Add(face, bbox)

    xmin, ymin, zmin, xmax, ymax, zmax = bbox.Get()
    if as_vec is False:
        return xmin, ymin, zmin, xmax, ymax, zmax
    else:
        return gp_Vec(xmin, ymin, zmin), gp_Vec(xmax, ymax, zmax)
```

It's the first working version I got. Do you want to pull that as an example? 
I think the ticket can be closed.

Thank you for your time!

jf---, 2016-10-11:
hi @maciekgroch , interesting, thanks for sharing this insight...

`get_boundingbox` is part of pythonocc-utils, but some thinking is required; perhaps supplying a topology traversal function makes sense as an argument

tpaviot, 2016-10-12:
@maciekgroch yes of course you can submit your code to be part of our example suite. The Topo class is available from the core_topology_traverse example

maciekgroch, 2016-10-12:
After all it doesn't work as expected. I played a bit with different methods and at some point I thought I had what I wanted:
```
from OCC.Bnd import Bnd_Box
from OCC.BRepBndLib import brepbndlib_Add
from OCC.gp import gp_Vec
from OCCUtils.Topology import Topo

def get_boundingbox(shape, tol=1e-6, as_vec=False):
    """ return the bounding box of the TopoDS_Shape `shape`
    Parameters
    ----------
    shape : TopoDS_Shape or a subclass such as TopoDS_Face
        the shape to compute the bounding box from
    tol: float
        tolerance of the computed boundingbox
    as_vec : bool
        wether to return the lower and upper point of the bounding box as gp_Vec instances
    Returns
    -------
        if `as_vec` is True, return a tuple of gp_Vec instances
         for the lower and another for the upper X,Y,Z values representing the bounding box
        if `as_vec` is False, return a tuple of lower and then upper X,Y,Z values
         representing the bounding box
    """
    bbox = Bnd_Box()
    bbox.SetGap(tol)
    topo = Topo(shape)
    for face in list(topo.faces()):
        brepbndlib_Add(face, bbox)

    xmin, ymin, zmin, xmax, ymax, zmax = bbox.Get()
    if as_vec is False:
        return xmin, ymin, zmin, xmax, ymax, zmax
    else:
        return gp_Vec(xmin, ymin, zmin), gp_Vec(xmax, ymax, zmax)

```

However, at first I used the following (based on your simple mesh example): 
```
import numpy as np
from OCC.BRep import BRep_Builder, BRep_Tool
from OCC.BRepMesh import BRepMesh_IncrementalMesh
from OCC.TopExp import TopExp_Explorer
from OCC.TopoDS import TopoDS_Compound, topods_Face
from OCC.TopAbs import TopAbs_FACE
from OCC.TopLoc import TopLoc_Location

def _calculate_bb_dimension(self, shape):
    BRepMesh_IncrementalMesh(shape, 0.8)
    builder = BRep_Builder()
    comp = TopoDS_Compound()
    builder.MakeCompound(comp)

    bt = BRep_Tool()
    ex = TopExp_Explorer(shape, TopAbs_FACE)
    positions = []
    while ex.More():
        face = topods_Face(ex.Current())
        location = TopLoc_Location()
        facing = (bt.Triangulation(face, location)).GetObject()
        tri = facing.Triangles()
        tab = facing.Nodes()
        for i in range(1, facing.NbTriangles() + 1):
            trian = tri.Value(i)
            index1, index2, index3 = trian.Get()
            for j in range(3):
                if j == 0:
                    m = index1
                    n = index2
                elif j == 1:
                    n = index3
                elif j == 2:
                    m = index2
                positions.append(tab.Value(m))
                positions.append(tab.Value(n))
        ex.Next()
    positions = map(lambda x: (x.X(), x.Y(), x.Z()), positions)
    min_xyz = np.array([min(positions, key=lambda x: x[i])[i] for i in range(3)])
    max_xyz = np.array([max(positions, key=lambda x: x[i])[i] for i in range(3)])
    dim_xyz = max_xyz - min_xyz
    min_xyz -= .03 * dim_xyz
    max_xyz += .03 * dim_xyz
    return min_xyz, max_xyz

```
Now the results:
If I execute just the first method the results are as the following:
```
-12.7900961718106,
 -6.60104293345751,
 -6.9371360356408704,
 12.789196858597599,
 8.408685875528969,
 6.68892024837101
```

then if the second method is executed:
```
-13.46589958, 
 -6.80504264,
  -0.80360052
13.48277862,
   8.7113905 , 
  6.89155951
```
And when the first method is executed after calling the second, the results are as the following:
```
-12.94688324556817,
 -6.6095804164526575,
 -0.8294950791375497,
 12.963762288762535,
 8.515928276315334,
 6.917454067292367
```

Should the shape be influenced by meshing it?

jf---, 2016-10-12:
`brepbndlib_Add` has a `useMesh` flag, right? so re-meshing the shape could indeed have an influence, when the mesh is explored to compute the boundingbox

maciekgroch, 2016-10-13:
Well, I don't like that design. IMHO if the flag `useMesh ` is set as `True` then it should be meshed _inside the method_ or a mesh should be passed to the method. Had you not told me how it worked, it would not be clear for me.
 
Nevertheless, thank you for your time. You both helped me a lot. Should I have any more issues I will contact you :).

jf---, 2016-10-13:
No worries @maciekgroch , good to have you on board 👍

tpaviot, 2017-05-04:
I added an example, using an optional meshing parameter, see commit 8463fc4

closed 2017-05-04 by tpaviot