tpaviot/pythonocc-core #341: Default SimpleGui uses a quarter of the available window

henrybaxter, 2016-10-17:
https://www.dropbox.com/s/f3jrbz2mze0hr8y/Screenshot%202016-10-16%2021.44.40.png?dl=0

This is on a retina display, and as I write this, I wonder if that's why? Anyway the following code gives the screenshot above:

    from OCC.Display.SimpleGui import init_display
    from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox

    display, start_display, add_menu, add_function_to_menu = init_display()
    my_box = BRepPrimAPI_MakeBox(10., 20., 30.).Shape()

    display.DisplayShape(my_box, update=True)
    start_display()

That is, the viewer only uses the bottom left quarter of the window.

Note the code above is copied from the "hello dumb box" tutorial.

I would be more than happy to provide more information or try to fix it if someone has an idea.

henrybaxter, 2016-10-17:
Also note that it scales with the resize value, so eg
    ... = init_display('qt-pyqt5', (200,200))
gives a smaller window, and still the viewer uses the lower left quarter.

henrybaxter, 2016-10-17:
Perhaps this bug (and fix) is related, using Graphic3d_RenderingParams::Resolution to 'display correctly fixed size elements on hidpi screens' https://tracker.dev.opencascade.org/view.php?id=25350 ?

henrybaxter, 2016-10-17:
I am unable to access any member of the V3d_View's Graphic3d_RenderingParams, eg

        p = self.View.RenderingParams()
        print(p.Resolution())
    AttributeError: 'Graphic3d_RenderingParams' object has no attribute 'Resolution'

Not sure what this would tell me if I could. I do know that the View_3d's Size() and the QMainWindow frameSize() both return the logical pixel size of the main window on the screen. I feel like the View_3d somehow has to find out it should render 2x bigger. Is that the View_3d's job, or some other object?

jf---, 2016-10-17:
hi @henrybaxter , so that fix you mention if for OCCT 7.0.1, while the version exposed in pythonocc corresponds to 6.8.0

I've dabbled with this problem myself.
I think the easier solution, well workaround might be to get the NSView from Qt, and set the correct arguments from that object.

`Cocoa_Window` [is constructed here](https://github.com/tpaviot/pythonocc-core/blob/master/src/Visualization/Display3d.cpp), so an alternative might be call the methods for hi-dpi there?

Hope that helps to get some grip on the issue...

henrybaxter, 2016-10-17:
@jf--- thank you so much for your response

To be honest I don't know what the correct arguments are for NSView - is the problem that the NSWindow object thinks it can do hidpi, but OCCT is writing as if it were low?

I'll keep looking, but if anybody has more hints, I'd really appreciate it!

jf---, 2016-10-17:
hi @henrybaxter ,

So IIRC, the quick & dirty way is to use `NSView's` method  `wantsBestResolutionOpenGLSurface` IIRC... [See this](http://steamcommunity.com/app/234650/discussions/0/648814843432299668/)

Things are more tricky though; imagine moving your app from a hi-dpi, to a traditional monitor ( attached to your laptop, a beamer, not so hypothetical ) 

Now you need to set the DPI according to this event... [more on the topic here](https://developer.apple.com/library/content/documentation/GraphicsAnimation/Conceptual/HighResolutionOSX/APIs/APIs.html)

So Qt5 has proper support for this, so it should be doable, when we add a method for setting the backend resolution...

Probably Qt5 comes with such an example, but havent checked...

OK, I hope that makes things more actionable?

jf---, 2016-10-17:
@henrybaxter , perhaps its possible to prototype this in [pyobjc](https://pypi.python.org/pypi/pyobjc)?
if you can find the correct NSView instance, then you're golden...

if you [look here](https://github.com/tpaviot/pythonocc-core/blob/master/src/addons/Display/qtDisplay.py#L108), `self.GetHandle()` is a pointer to either the NSView or NSWindow. 

Its quite possible that this is your hook to solving this, without delving into C++, though that's probably the cleaner / easier route. But who knows, perhaps If black magic is your thing 😉

jf---, 2016-10-28:
I think I have a solution:

```python
                import objc
                nsview = objc.objc_object(c_void_p=self.GetHandle())
                nsview.setWantsBestResolutionOpenGLSurface_(True)
```

i guess after that we need to re-scale the dimensions of the view

jf---, 2016-10-28:
OK, this seems to work out, while initializing the view:

```python

    @pyqtSlot()
    def InitDriver(self):
        try:

            _nsview_pointer = self.GetHandle()
            screen = QGuiApplication.instance().primaryScreen()
            pixelDensity = screen.physicalDotsPerInch()

            if pixelDensity >= 147:

                try:
                    import objc
                except ImportError:
                    log.critical("no PyObjc module for retina")
                else:
                    nsview = objc.objc_object(c_void_p=_nsview_pointer)
                    nsview.setWantsBestResolutionOpenGLSurface_(True)
                    scaleFactor = nsview.backingScaleFactor()
                    rec1 = screen.geometry()
                    nsview.setBounds_(
                        ((rec1.x(), rec1.y()), (rec1.width()*scaleFactor, rec1.height()*scaleFactor))
                    )

            self._display = OCCViewer.Viewer3d(self.GetHandle())
            self._display.Create()

```

nyholku, 2017-01-07:
I'm experiencing this problem also.

So it looks like this has been solved at some level and merged.

I'm very new to Python (but an old hand in programming in C/C++/Java) so this maybe a very basic question:

How do I use this? 

I'm just dipping my toes into the `pythonocc` water and so far I've installed `pythonocc ` with `conda` and trying things out in Eclipse and things have 'just worked'. 

So how do I 'install' these changes?

tpaviot, 2017-01-07:
@nyholku There are nightly builds for pythonocc if you want to benefit from recent changes that have not been released yet. To install with conda:

````
conda install -c tpaviot pythonocc-core=nightly
````

This installs latest master build (pythonocc-0.17.2-dev).

nyholku, 2017-01-07:
Brilliant, this is exactly what I was hoping for! 

Except that when I try it I get:

~~~
nyholku$ conda install -c tpaviot pythonocc-core=nightly
Fetching package metadata .........
Solving package specifications: .

PackageNotFoundError: Package not found: '' Package missing in current osx-64 channels: 
  - pythonocc-core nightly*

You can search for packages on anaconda.org with

    anaconda search -t conda pythonocc-core

nyholkus-MacBook-Pro-5:cmake-build nyholku$ anaconda search -t conda pythonocc-core
Using Anaconda API: https://api.anaconda.org
Run 'anaconda show <USER/PACKAGE>' to get more details:
Packages:
     Name                      |  Version | Package Types   | Platforms      
     ------------------------- |   ------ | --------------- | ---------------
     DLR-SC/pythonocc-core     |     0.17 | conda           | linux-64, win-32, win-64, osx-64
                                          : A python wrapper for the OCE library
     jf/pythonocc-core         |   0.16.5 | conda           | osx-64         
                                          : python bindings for OCE - OpenCASCADE Community Edition
     pythonocc/pythonocc-core  |   0.17.1 | conda           | linux-64, win-32, win-64, osx-64
                                          : A python wrapper for the OCE library
     tpaviot/pythonocc-core    |  nightly | conda           | linux-64, win-32, win-64, osx-64
Found 4 packages
nyholkus-MacBook-Pro-5:cmake-build nyholku$ 
~~~

nyholku, 2017-01-07:
FYI: I accidentally (!) tried this with Qt4 and the 1/4 screen problem is not present with Qt4.

tpaviot, 2017-01-08:
I fixed the nightly-builds on anaconda, @nyholku can you please retry
```
conda install -c tpaviot pythonocc-core=nightly
```
and report success/failure

nyholku, 2017-01-08:
Hi  @tpaviot , the nighly build now works for me, great, thank you very much!

Unfortunately this does not fix the issue #341 i.e. the display still uses just the lower left quarter of the window, I was hoping that the nightly build would fix that. Maybe I misread what it was said on that ticket.

tpaviot, 2017-01-08:
@nyholku thanks for thee feedback

@jf--- I thought your retina patch solved this issue, didn't it ?

jf---, 2017-01-08:
@tpaviot yes, we could make it pyqt5 specific?

nyholku, 2017-01-08:
@tpaviot , @jf--- 

Can you clarify this a bit, I was under the impression that I'm running Qt5, so this patch should have done it for me from the nightly build?

This what I have:
~~~
conda list
# packages in environment at /Users/nyholku/miniconda3:
#
anaconda-client           1.6.0                    py35_0  
clyent                    1.2.2                    py35_0  
conda                     4.2.13                   py35_0  
conda-env                 2.6.0                         0  
freeimageplus             3.17.0                        0    dlr-sc
freetype                  2.5.5                         1  
gl2ps                     1.3.8                         0    dlr-sc
icu                       54.1                          0  
libpng                    1.6.27                        0  
oce                       0.17.2                        3    dlr-sc
openssl                   1.0.2j                        0  
pip                       8.1.2                    py35_0  
pycosat                   0.6.1                    py35_1  
pycrypto                  2.6.1                    py35_4  
pyqt                      5.6.0                    py35_1  
python                    3.5.2                         0  
python-dateutil           2.6.0                    py35_0  
pythonocc-core            nightly                 py35_25    tpaviot
pytz                      2016.10                  py35_0  
pyyaml                    3.12                     py35_0  
qt                        5.6.2                         0  
readline                  6.2                           2  
requests                  2.11.1                   py35_0  
ruamel_yaml               0.11.14                  py35_0  
setuptools                27.2.0                   py35_0  
sip                       4.18                     py35_0  
six                       1.10.0                   py35_0  
sqlite                    3.13.0                        0  
tbb                       4.3.6                         0    dlr-sc
tk                        8.5.18                        0  
wheel                     0.29.0                   py35_0  
xz                        5.2.2                         0  
yaml                      0.1.6                         0  
zlib                      1.2.8                         3  
[1]+  Terminated: 15          python
~~~

rainman110, 2017-03-20:
Hey guys. I know this problem cause I already ran across it. It is a issue of OpenCASCADE 6.7.0 ( 6.9.0?) and Qt5. 

There is also a issue in the OCCT mantis: 
https://tracker.dev.opencascade.org/view.php?id=26603
https://tracker.dev.opencascade.org/view.php?id=27922

This can be circumvented, by using the OcctWindow class, which handles resizing correctly. This class is part of the OCCT Qt example since 7.0.0. This class must be integrated into the pythonocc qt viewer though.

tpaviot, 2020-01-05:
Please try with latest release (7.4.0rc1), it should be fixed. Reopen the ticket if ever it still fails.

closed 2020-01-05 by tpaviot