tpaviot/pythonocc-core #342: Expose OCCT 7.0.1 in pythonocc

henrybaxter, 2016-10-17:
For the ultimate fix to #341 @jf--- you mention exposing 7.0.1 instead of 6.8.

I would like to work on a branch that does just that, but I will need some guidance.

First, isn't python-occ actually exposing OCE? In which case, https://github.com/tpaviot/oce seems to be tracking OCCT 6.8. In particular https://groups.google.com/forum/#!topic/oce-dev/kaVarqjggBQ here @tpaviot talks about the work involved.

So is my first step to help the OCE effort to move to 7.0.1?

jf---, 2016-10-17:
hi @henrybaxter , 

so maintaining the python wrapper for OCC is a gargantuan task.
So if you want to provide support for hi-dpi `Visualization.cpp` is where to look...
You could simple add a C++ method for handling resolution. 
This is most definitely the quick fix.

That said, for porting pythonocc to 7.0.1, its best to have that discussion at [pythonocc-generator](https://github.com/tpaviot/pythonocc-generator), which is where the code is that generates the OCE bindings.

To delve into this matter, @tpaviot did [a great write up](https://github.com/tpaviot/pythonocc-generator/blob/master/Documentation.md), such to have more of an idea what it encompasses.

henrybaxter, 2016-10-17:
@jf--- thank you for the information, I read what @tpaviot wrote there, though it is conspicuously missing the part titled "Making your own wrapper - This section describes how to upgrade the wrapper for an oce version change", I do understand roughly what's happening with SWIG here.

I will pursue helping with the upgrade at https://github.com/tpaviot/pythonocc-generator

As for the faster fix, I will post here if and when I figure it out!

jf---, 2016-10-17:
Awesome @henrybaxter , looking fwd, I share a similar itch, this would be an excellent contrib...

tpaviot, 2016-10-18:
Next step is rather To support occt693. Experimenting with occt7 is interesting feel free to create a branch for testing

tpaviot, 2019-12-29:
Deprecated. Current release wraps occt 740

closed 2019-12-29 by tpaviot