tpaviot/pythonocc-core #348: meta issue release 0.17.1

jf---, 2016-10-31:
lets discuss to define issues to be resolved for **0.17.1**
here's something to start off with, in no particular order:

## Priority:

- #139 

## Positive:

- #5, #312  ( glsl support )
- #260 ( clipping 3d models in OpenGL )
- #346 ( incl assimp support ), related #296, related #185 
- #194 ( unicode in file name / path )
- #301 ( camera )

---

## Perhaps:

- #326 
- #337 
- #341 
- #289, related #307  ( this is key competence for pythonocc, but this is a sub-project, that requires someone other than the core-maintainers to champion it; eg urgent matter but low priority )

---

## Close?

- #279 possible related #139 
- #308 
- #300

jf---, 2016-10-31:
suggested division of labour:

- I'm having a pretty good grip on things on anything concerning graphics and Qt
  - except where wrappers are showing their limits ( some learning to do! )
  - #139, #5, #260, #346 , #301 
- also support for meshes

tpaviot, 2016-10-31:
@jf--- After the 0.17 big release, let's proceed with smaller steps. After brep font is merged, I will push osmething related to webgl and it will be time for 0.17.1 (I need the webgl stuff asap). This issue can renamed as 0.17.x roadmap, with different items 0.17.1, 0.17.2 etc.

As the work goes on the python part, there also should be an effort to release oce-0.18, update the wrapper for 0.18. Indeed most pythonocc work is 0.17/0.18 independant (except maybe the fuzzy bool ops that require 0.18).

jf---, 2016-10-31:
@tpaviot , curious about the webgl...
ok, lets set smaller steps, that makes sense.

- lets agree that 0.17.1 can basically be shipped after your webgl effort?

can i suggest a graphics related 0.17.2?

- there's a dangling PR #314 and an important unresolved issue #139, and I'm really eager to get #5 done... ( this could be 0.17.2 )

- 0.17.3 `AIS_Triangulation` / assimp support ( possible in pythonocc-contrib? )

tpaviot, 2017-05-04:
0.17.3 was released.

Next step is to support oce-0.18

closed 2017-05-04 by tpaviot