tpaviot/pythonocc-core #352: Travis build broken

tpaviot, 2016-11-08:
The last successfull build of the master branch is 4 days old. Since that, compilation fails for python2.7 with a curious exception (see https://travis-ci.org/tpaviot/pythonocc-core/jobs/172543060):
```
compiling .pyc files...
Compiling /home/travis/miniconda/envs/_build/lib/python2.7/site-packages/pip/_vendor/html5lib/treewalkers/etree.py ...
  File "/home/travis/miniconda/envs/_build/lib/python2.7/site-packages/pip/_vendor/html5lib/treewalkers/etree.py", line 13
    from pip._vendor. import string_types
                           ^
SyntaxError: invalid syntax
```
It only affects py2.7 and Travis. On appveyor, everything works fine. Hard to say if that comes from Travis or conda. @rainman110 Are you aware of any change on the conda side that coud explain this regression ?

rainman110, 2016-11-08:
@tpaviot No, I have no clue. Our conda packages are also unchanged for more than a month now. It must be some conda updates from anaconda that lead to this regression.

Look at the dot behind vendor... This looks like the cause of it.

tpaviot, 2016-11-08:
Similar issue 5 days ago here https://github.com/pypa/pip/issues/4059

tpaviot, 2016-11-08:
ok I got it it's an issue with latest pip version on python 2.7. On my windows machine, python2.7, upgrading pip leads to the same problem:
```
    File "[...]\appdata\local\temp\pip-build-kkpnjo\pip\pip\_vendor\ht
ml5lib\treewalkers\etree.py", line 13
      from pip._vendor. import string_types
                             ^
  SyntaxError: invalid syntax

Cleaning up...
Exception:
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\pip\basecommand.py", line 215, in main
    status = self.run(options, args)
  File "C:\Python27\lib\site-packages\pip\commands\install.py", line 317, in run
[...]
```

tpaviot, 2016-11-08:
It seems it was fixed in a 9.0.1 pip relase published on the 6th of november. The fix is here https://github.com/pypa/pip/pull/4068/commits/d884c306b6a8ed39b348a3cd30cbba6cad560acf

tpaviot, 2016-11-08:
Unfortunately the version 9.0.1 is not available yet in the anaconda channel (still 9.0.0).

rainman110, 2016-11-08:
You could also force an older version of pip .

tpaviot, 2016-11-08:
yep that's what I did see 84b2aec2b6cff36fdc571c70d85a9c2d1d765f26 but it does not seem necessary. Indeed pip 9.0.1 is installed on travis, maybe they updated the channel. I triggered another build of master to check if it's ok

closed 2016-11-08 by tpaviot