tpaviot/pythonocc-core #360: Python 3 compatibility

kngbuzzo, 2016-11-22:
Hello everyone,
recently, I have been trying to use pythonocc in Python 3 (I use the conda packages for Windows provided by the dlr-sc channel). Specifically, I tried to run the example script I created for issue #308 found [here](https://github.com/kngbuzzo/pythonocc_examples) in a Python 3 environment. However, on my machine, I get this error message:

```
Traceback (most recent call last):
  File "pythonocc_ellipsoid_example.py", line 149, in <module>
    intersect(basis_profiles, v)
  File "pythonocc_ellipsoid_example.py", line 116, in intersect
    occsurf = get_pythonocc_bspline_surface(surf_input)
  File "pythonocc_ellipsoid_example.py", line 66, in get_pythonocc_bspline_surface
    return Geom_BSplineSurface(cpts, uknots, vknots, umult, vmult, udeg, vdeg, uperiod, vperiod).GetHandle()
  File "C:\Anaconda2\envs\py3\lib\site-packages\OCC\Geom.py", line 5518, in __init__
    _Geom.Geom_BSplineSurface_swiginit(self, _Geom.new_Geom_BSplineSurface(*args))
TypeError: in method 'new_Geom_BSplineSurface', argument 2 of type 'TColStd_Array2OfReal const &'
```

In Python 2 no error occurs. I used Anaconda 4.1.1 and pythonocc 0.17 each time. Do you have any idea what might be causing this? Is pythonocc even meant to be used with Python 3?

Regards

jf---, 2016-11-22:
wrt py3k compatibility, yes pythonocc supports py3k

kngbuzzo, 2016-11-22:
Ok, I figured it out. I ended up in the wrong incarnation of the constructor, which expects a weights array ([see here](https://www.opencascade.com/doc/occt-6.9.0/refman/html/class_geom___b_spline_surface.html#a2cceaf7bab6ce3a234166ebbd9671f6e); the second one instead of the first - thus the missing TColStd_Array2OfReal input). As it turns out, this was due to the udeg and vdeg parameters not being interpreted correctly as integers, so you were right @jf--- . All I needed to do to solve this was explictly convert them. Kinda weird that the Pythons behave this differently, but there is probably not much you can do about that.

jf---, 2016-11-23:
Thanks for the feedback @kngbuzzo 
Could you close the issue please?

closed 2016-11-23 by kngbuzzo