tpaviot/pythonocc-core #361: Travis Issue libcurl-glic

tpaviot, 2016-11-23:
There might have been an update to travis workers. Indeed linux builds don't passa nymore because of a failure related to libcurl. See https://travis-ci.org/tpaviot/pythonocc-core/jobs/177998481

@rainman110 Did you change something to your curl build ?

rainman110, 2016-11-23:
@tpaviot No, i have not changed anything to libcurl. If you look closely, you can see that the new job is using cmake 3.6 now. Before it was cmake 3.3.1 . The new cmake requires curl and installs it.

I suppose, i compiled curl on a too new gcc / glibc. I can try to fix this. As a workaround, force the old cmake 3.3.1.

closed 2016-11-24 by tpaviot

rainman110, 2016-11-24:
@tpaviot I changed now libcurl to require only glibc 2.14. So in theory, your workaround can be reverted now.

tpaviot, 2016-11-24:
@rainman110 done, that works. Thank you for both the tip and the quick update

PR #362 (2016-11-24): force cmake to version 3.3.1 in conda meta to prevent glibc travis issue
Fix #361