tpaviot/pythonocc-core #369: Return python float from math_Matrix.Value()

trelau, 2016-12-17:
The math_Matrix.Value() method should return a python float rather than a pointer to a c++ double.

From Thomas...

A workaround is to:
-open /src/SWIG_files/wrapper/math.i
-change line 3131 from:
    Standard_Real & Value (const Standard_Integer Row,const Standard_Integer Col);
to:
    Standard_Real Value (const Standard_Integer Row,const Standard_Integer Col);
(just remove the &)

-rebuild/install pythonocc

jf---, 2016-12-17:
that's a nice find @trelau , I guess we'll need to check whether the patterns occurs in the wrapper.
since wrappers are generated automagically, we'll need to check.

thanks for the report...

tpaviot, 2016-12-17:
Personal reminder: in the pythonocc-generator script, on line https://github.com/tpaviot/pythonocc-generator/blob/master/src/generate_wrapper.py#L891, the n parameters case has to be handled.

tpaviot, 2019-01-10:
Fixed in current master

closed 2019-01-10 by tpaviot