tpaviot/pythonocc-core #371: can't push Graphic3d_Shader variables

jf---, 2016-12-18:
```python
from OCC.Graphic3d import (Graphic3d_ShaderProgram, Graphic3d_TOS_VERTEX, Graphic3d_TOS_FRAGMENT,
                           Graphic3d_ShaderObject)
from OCC.TCollection import TCollection_AsciiString

# fragment shader
fs = """
void main(void)
{
    gl_FragColor=vec4(0.0, 1.0, 0, 1.0);
}
"""

# vertex shader
vs = """
void main()
{
    gl_Position = occProjectionMatrix * occWorldViewMatrix * occModelWorldMatrix * occVertex;
}
"""

# construct the shader, load, compile and attach the GLSL programs
vs_shader = Graphic3d_ShaderObject.CreateFromSource(Graphic3d_TOS_VERTEX, TCollection_AsciiString(vs))
fs_shader = Graphic3d_ShaderObject.CreateFromSource(Graphic3d_TOS_FRAGMENT, TCollection_AsciiString(fs))
aProgram = Graphic3d_ShaderProgram()
aProgram.AttachShader(fs_shader)
aProgram.AttachShader(vs_shader)

time_ = 0.0

```

The issue is where:

```
# missing method, possible due to usage of templates?aProgram.PushVariable(TCollection_AsciiString("time), time_+=1)

```

tpaviot, 2016-12-18:
To access the PushVariable method, the Graphic3d.i swig file must be modified the following way:
```
template<class T> Standard_Boolean PushVariable (const TCollection_AsciiString & theName,const T & theValue);
%template(PushVariableInt) PushVariable<int>;
%template(PushVariableFloat) PushVariable<float>;
```
After that, it's possible to use the method using:
```
time_+=1
aProgram.PushVariableFloat(TCollection_AsciiString("time"), time_)
```
All the possible types must be defined at the swig level, no way to do it on the fly. However, datatypes string, float, int, bool should cover most cases (?).

This has to be automated in the generator, there maybe other templated methods.

jf---, 2016-12-18:
terrific, thanks for the feedback...