tpaviot/pythonocc-core #372: RuntimeError on self.Context.OpenLocalContext()

jf---, 2016-12-20:
```python
    def SetSelectionMode(self, mode=None):
        self.Context.CloseAllContexts()
>>>             self.Context.OpenLocalContext()
>>>             raises a RuntImeError
>>>             hit "G" in the viewer
>>>             I'm seeing this on py 3.5
        topo_level = next(modes)
        if mode is None:
            self.Context.ActivateStandardMode(topo_level)
        else:
            self.Context.ActivateStandardMode(mode)
        self.Context.UpdateSelected()
```

tpaviot, 2016-12-22:
@jf--- I can't reproduce the issue on Windows running latest master. Hiting the "G" key works as expected: the selected type object change from Vertex to Edge and Face and so on

jf---, 2016-12-22:
@tpaviot OK thanks, I'll need to dig deeper...

closed 2017-05-04 by tpaviot