tpaviot/pythonocc-core #376: SimpleGui and PyQt4

tpaviot, 2016-12-22:
When resizing the window:
```
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\OCC\Display\qtDisplay.py", line 215, in re
sizeGL
    self.setupViewport(width, height)
AttributeError: 'qtViewer3d' object has no attribute 'setupViewport'
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\OCC\Display\qtDisplay.py", line 215, in re
sizeGL
    self.setupViewport(width, height)
AttributeError: 'qtViewer3d' object has no attribute 'setupViewport'
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\OCC\Display\qtDisplay.py", line 215, in re
sizeGL
    self.setupViewport(width, height)
AttributeError: 'qtViewer3d' object has no attribute 'setupViewport'
```

jf---, 2016-12-22:
hi @tpaviot  this is addressed in PR #375

tpaviot, 2017-01-01:
Fixed by commit 2a9210f

closed 2017-01-01 by tpaviot