tpaviot/pythonocc-core #378: Qt5 error on Windows using pythonocc-0.17.1

tpaviot, 2017-01-01:
pythonocc-0.17.1, issue with Qt5 on Windows:
````
(pyocc-0.17-py2) D:\Devel\pythonocc-core\examples>python core_helloworld.py
INFO:OCC.Display.backend:backend loaded: qt-pyqt5
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt5
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
WNT window created.
Traceback (most recent call last):
Traceback (most recent call last):
  File "core_helloworld.py", line 21, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "Miniconda2\envs\pyocc-0.17.1-py2\lib\site-packages\OCC\Display\SimpleGui.py", line 201, in init_display
    win.canva.InitDriver()
  File "Miniconda2\envs\pyocc-0.17.1-py2\lib\site-packages\OCC\Display\qtDisplay.py", line 136, in InitDriver
    self._display.Create()
  File "Miniconda2\envs\pyocc-0.17.1-py2\lib\site-packages\OCC\Display\OCCViewer.py", line 137, in Create
    self.Init(self._window_handle)
RuntimeError: Aspect_GraphicDeviceDefinitionError
OpenGl_Window::CreateWindow: SetPixelFormat failed. Error code: 2000
````

tpaviot, 2017-01-01:
Note: this is a regression, everything used to work on 0.17

tpaviot, 2017-01-01:
Fixed by commit 2a9210ff0cbdf597fd25a0d73aae1d0b4b1e54b3

closed 2017-01-01 by tpaviot

HensonWong, 2019-12-20:
Hello Thomas, I want to ask a question, how can I integrate the window of pyocc into the window of pyqt5, this problem has troubled me for a long time, I hope to find the answer here