tpaviot/pythonocc-core #382: Compile fails on Macos 10.10.5 (Yosemite)

nyholku, 2017-01-07:
I tried to compile the `pythonocc` as follows (see transcript) and it seems that CMake config works fine but the compilation fails.

My prerequisites are as follows:

```
nyholku$ brew info freetype
freetype: stable 2.7.1 (bottled)

nyholku$ brew info oce
homebrew/science/oce: stable 0.17.2 (bottled)

brew info cmake
cmake: stable 3.7.1 (bottled), HEAD

 brew info swig
swig: stable 3.0.11 (bottled)
```

My freetype seems to be a bit newer than specified, is that a problem?

Compilation transcript:

```
nyholku$ git clone git://github.com/tpaviot/pythonocc-core.git
Cloning into 'pythonocc-core'...
remote: Counting objects: 6329, done.
remote: Compressing objects: 100% (21/21), done.
remote: Total 6329 (delta 6), reused 0 (delta 0), pack-reused 6308
Receiving objects: 100% (6329/6329), 17.92 MiB | 1.14 MiB/s, done.
Resolving deltas: 100% (4961/4961), done.
Checking connectivity... done.
nyholkus-MacBook-Pro-5:~ nyholku$ cd pythonocc-core
nyholkus-MacBook-Pro-5:pythonocc-core nyholku$ mkdir cmake-build
nyholkus-MacBook-Pro-5:pythonocc-core nyholku$ cd cmake-build
cmake -DOCE_INCLUDE_PATH=/usr/local/Cellar/oce/0.17.2_1/include/oce -DOCE_LIB_PATH=/usr/local/Cellar/oce/0.17.2_1/lib ..
-- The C compiler identification is AppleClang 7.0.2.7000181
-- The CXX compiler identification is AppleClang 7.0.2.7000181
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
--  
-- Found PythonInterp: /Users/nyholku/miniconda3/bin/python (found version "3.5.2") 
-- Found PythonLibs: /Users/nyholku/miniconda3/lib/libpython3.5m.dylib (found version "3.5.2") 
-- Build 64bit
-- Found SWIG: /usr/local/bin/swig (found version "3.0.10") 
-- Found Freetype: /usr/local/lib/libfreetype.dylib (found version "2.7.0") 
-- Platform:Darwin
-- output_path: Darwin/x86_64-MinSizeRel-64
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/nyholku/pythonocc-core/cmake-build
nyholkus-MacBook-Pro-5:cmake-build nyholku$ mak
-bash: mak: command not found
nyholkus-MacBook-Pro-5:cmake-build nyholku$ make
[  0%] Swig source
Scanning dependencies of target _XCAFPrs
[  0%] Building CXX object CMakeFiles/_XCAFPrs.dir/Darwin/x86_64-MinSizeRel-64/XCAFPrsPYTHON_wrap.cxx.o
[  0%] Linking CXX shared module Darwin/x86_64-MinSizeRel-64/_XCAFPrs.so
ld: warning: path '/usr/local/lib/libfreetype.dylib' following -L not a directory
[  0%] Built target _XCAFPrs
[  0%] Swig source
[... 221 lines cut ...]
      _wrap_TDataXtd_Constraint_GetType(_object*, _object*) in TDataXtdPYTHON_wrap.cxx.o
  "TDataXtd_Constraint::GetPlane() const", referenced from:
      _wrap_TDataXtd_Constraint_GetPlane(_object*, _object*) in TDataXtdPYTHON_wrap.cxx.o
  "TDataXtd_Constraint::GetValue() const", referenced from:
      _wrap_TDataXtd_Constraint_GetValue(_object*, _object*) in TDataXtdPYTHON_wrap.cxx.o
  "TDataXtd_Constraint::Inverted() const", referenced from:
      _wrap_TDataXtd_Constraint_Inverted__SWIG_1(_object*, int, _object**) in TDataXtdPYTHON_wrap.cxx.o
  "TDataXtd_Constraint::IsPlanar() const", referenced from:
      _wrap_TDataXtd_Constraint_IsPlanar(_object*, _object*) in TDataXtdPYTHON_wrap.cxx.o
  "TDataXtd_Constraint::Reversed() const", referenced from:
      _wrap_TDataXtd_Constraint_Reversed__SWIG_1(_object*, int, _object**) in TDataXtdPYTHON_wrap.cxx.o
  "TDataXtd_Constraint::Verified() const", referenced from:
      _wrap_TDataXtd_Constraint_Verified__SWIG_0(_object*, int, _object**) in TDataXtdPYTHON_wrap.cxx.o
  "vtable for TDataXtd_HArray1OfTrsf", referenced from:
      _wrap_new_TDataXtd_HArray1OfTrsf__SWIG_0(_object*, int, _object**) in TDataXtdPYTHON_wrap.cxx.o
      _wrap_new_TDataXtd_HArray1OfTrsf__SWIG_1(_object*, int, _object**) in TDataXtdPYTHON_wrap.cxx.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [Darwin/x86_64-MinSizeRel-64/_TDataXtd.so] Error 1
make[1]: *** [CMakeFiles/_TDataXtd.dir/all] Error 2
make: *** [all] Error 2
nyholkus-MacBook-Pro-5:cmake-build nyholku$ 

```

nyholku, 2017-01-07:
Just to add that my goal is *NOT* to compile pythonocc per se but to be able to enjoy the latest fixes. At this point I would very much like to see fix for the issue #341 work for me.

So if the nightly builds full fill this and I can install them then this is not a priority as far as I'm concerned. But if this is a real issue with build don't let this stop you ;)

ArdooTala, 2018-09-10:
Hi,
I've got the same problem and I can't find a fix to it. Did you find a fix?

tpaviot, 2019-01-11:
You need at least oce-0.18 or higher

closed 2019-01-11 by tpaviot