tpaviot/pythonocc-core #389: Jupyter notebook embedded rendering.

p-chambers, 2017-01-20:
I realise there is already an issue open for notebook rendering (#164), and I am already using the Tornadoweb renderer submitted by @rainman110 on the pythonocc-contrib repo. However, it's increasingly my desire to have the javascript output embedded in the notebook, rather than rendered in a TornadoWeb server and displayed through a port. 

I use pythonocc for an aircraft geometry library ([occ_airconics](https://github.com/p-chambers/occ_airconics)), and it would be a really nice feature if uploading my notebooks to the web would retain and render the geometry, as done by [openmodes](https://github.com/DavidPowell/OpenModes) in this [example](http://nbviewer.jupyter.org/github/DavidPowell/openmodes-examples/blob/master/How%20to%20create%203D%20plots.ipynb).

Is this something you guys might eventually work on?

tpaviot, 2017-12-17:
@p-chambers I just discovered the pythreejs project (https://github.com/jovyan/pythreejs @jovyan). They just a released a new version. I do think it could be a perfect candidate to the basis of a jupyter renderer for pythonocc, using threejs. I quickly checked pythreejs feature in a jupyter notebook , it's just amazing. I started to work on an experimental implementation for pythonocc, based upon:

* an addtion to the Tesselator.cpp code, in order to be able to get the vertex coordinates as a list of lists
[[x1, y1, z1], [x2, y2, z2,] .... ] --  c++ work :(
* a BufferGeometry instance (see https://github.com/jovyan/pythreejs/blob/master/examples/BufferAttributes%20and%20BufferGeometry.ipynb) and pass this list 
* a basic renderer.

As pythreejs relies on numpy, it should be possible, in a second stage, to extend the Tesselator class to return numpy ndarrays, in order to speed up the rendering process. This is related to issue #185.

I will update this ticket as the work is going on. 

ping @jf--- @rainman110 @aothms for information

aothms, 2017-12-17:
@tpaviot I may have time to pitch in here and there as this is also highly on my todo list!

> c++ work

Out of curiosity why does this need to be in C++? Perhaps you can also slice the lists in Python? If the numpy work is done first, then reshaping the data is effortless.

> As pythreejs relies on numpy, it should be possible, in a second stage, to extend the Tesselator class to return numpy ndarrays

As @rainman110 mentioned in the other thread, I am not sure if this is the only way forward, possible discrepancies between numpy versions can make this increasingly complex (although probably the basic interface is rather stable).

Perhaps calling `numpy.frombuffer()` in the python code would be easier. You can pass it a string/bytes object or perhaps a python `array.array` that you create in C. You remove the need of an additional dependency and additional api to talk to. It is just as efficient I think.

Let me know if there is somewhere I can pitch in (preferably pure python as I don't have a development environment or experience setup for pyocc c++).

tpaviot, 2017-12-17:
@aothms Only a part of the work can be done in c++. Currently, my Tesselator class does not allow me to get the vertex coordinates list (as a list of floats for instance). When it's achieved, I can of course use either python slice or numpy reshape. The numpy work would be at the python level, not at the c++ level, thus there should not be any problem.

After that, in a second stage, an "optimization" work could lead to move the numpy work at the c++ level, but I'm not sure it could allow to save memory/computer time, it's just a guss coming from the discussion of issue #185.

@aothms I'll let you know when you can download a new binary using conda (my travis and appveyor test builds automatically uploads conda packages to my tpaviot anaconda channel so that everyone can quickly test new developments)

tpaviot, 2017-12-17:
commits 91c0dc8e3cfe75fc43e87c6f5be8e44c798291b6 and  4fbccdc22c7c5ec4ec6322a31d8ddd14f499725d allows to get the vertex coordinates as a numpy array

tpaviot, 2017-12-21:
@aothms making good progresses ! Still have to deal with face normals, I will sens you a link before tomorrow so that you can test/improve if you want.

[image]

aothms, 2017-12-22:
Wonderful! If it's not there yet, one thing I'd hope to contribute is rendering edges, but I'll probably need to extend your C++ tesselator for that as well.

tpaviot, 2017-12-22:
Edge rendering is already supported in x3dom and threejs pythonocc renderers. That should not be that hard to implement using pythreejs. Here is another screenshot from the latest commit of the review/tess branch

[image]

Live demo at http://bit.ly/2l0AElZ

I did not succeed to export face normals, it will be possible in the next pythreejs, unfortunately I fail installing the latest dev version.

aothms, 2017-12-27:
> unfortunately I fail installing the latest dev version.

I managed to get it working (on ubuntu). From the log I saw you attached to a ticket you opened there it seems you need to have `npm` installed (and added to PATH) even for the `pip install` step.

tpaviot, 2017-12-27:
I do not use pip to install pythreejs, I don't manage to get something working. So far, I'm working with the latest stable pythreejs release (0.4) available through conda:
``` conda install -c conda-forge pythreejs=0.4.1```

Unfortunately, not tested release 1.0.0.beta yet.

tpaviot, 2017-12-27:
And yes, npm has to be installed if you want to install pythreejs using pip. On my linux/windows machine, pip install works, but I fail installing the jupyter notbook extension.

tpaviot, 2018-01-25:
The jupyter rendering feature is now part of the current master branch

closed 2018-01-25 by tpaviot