tpaviot/pythonocc-core #390: Export to image / FreeImage not working

dkaripidescp, 2017-01-20:
I'm seeing an issue exporting to images in any other format other than PPM. Here's are steps that reproduce the problem for me:

Installed pythonocc via the conda binaries.
Downloaded the pythonocc-1.72 source to get the examples directory.
Ran the example "core_display_export_to_image.py"
Export an image in any of the 4 formats
The image file will have the right extension (.png, .jpeg, and so on) but will actually be in PPM format.

There is a comment in core_display_export_to_image.py that reads:
```
# in order to be able to export to usual file formats
# oce must have been compiled with FreeImage
# otherwise, images will be exported to PPM format
```
Since I am using the conda binaries I am under the impression that oce has been compiled with FreeImage. 

As a final test I installed/compiled all the other prerequisites, compiled oce with FreeImage configured to ON, and compiled pythonocc-core manually. I get the same errant behavior.

tpaviot, 2017-01-21:
@dkaripidescp thanks for the report, I'm investigating. This maybe an oce issue.

tpaviot, 2017-01-21:
@dkaripidescp There seems to be an issue with oce compilation. Although you have downloaded all required dependencies, and compiled oce with FreeImaged configure to ON, you still have to manually define the FREEIMAGE_INCLUDE_DIR. Indeed if oce does not find the freeimage library, there isn't any warning or log that tells you it's not found. And oce compile without freeimage.

If you can rebuild oce on your machine and try with setting FREEIMAGE_INCLUDE_DIR, please report success/failure. I've been trying to update oce builds on anaconda.

tpaviot, 2017-01-22:
@dkaripidescp ok, that's what I thought. I rebuilt a proper oce and uploaded the binary to conda. You can test with this other oce binary. First remove your oce and pythonocc install:
```
$ conda remove oce
```
then install the new oce build from the oce repository (still experimental though)
```
$ conda install -c oce oce=0.17.2
```
and update your pythonocc-core copy
```
$ conda install -c pythonocc pythonocc-core
```
you may have to add conda-forge to your channels.

tpaviot, 2017-04-24:
Fixed.

closed 2017-04-24 by tpaviot