tpaviot/pythonocc-core #392: Fail to install with conda

trelau, 2017-01-21:
I've used conda in the past with success, but now a bunch of strange messages pop up and i get ImportError: DLL load failed. I'm on Python 3.5 Win64

Here is the conda output...

[image]

and here is the error when running python

[image]

In an attempt to compile pythonocc w/ OCE 0.18.0 I removed my current installation of pythonocc and now I can't get it back...lesson learned....

tpaviot, 2017-01-22:
@trelau wow, such a mess, I've never seen that before. do you use different and specific conda environments ? There should not be any problem installing/removing packages when working with environments.
You can always go back to a working installation by creating a new environment and install packages.

trelau, 2017-01-22:
I tried a clean new install in a new environment using:

conda create -n pythonocc -c pythonocc -c dlr-sc pythonocc-core==0.17.2 python=3

The same strange messages showed up at the end, BUT...the same messages showed up after installing numpy and scipy in that new environment....so perhaps it is not a pythonocc issue...maybe an issue in a conda update? Things seem to get installed though...

I noticed when I installed in a new environment it installed the OCE libs/dll's and other associated libraries. When I was doing the install on my main environment the libs/dll's (TK*) did not show up like before, hence the ImportError as shown above.

tpaviot, 2017-05-21:
@trelau Any progress regarding this issue ?

trelau, 2017-05-21:
@tpaviot No. I still get the error messages after a number of conda updates and for packages other than pythonocc. Everything always seem to install ok, so I've just looked the other way. I don't think it is a pythonocc issue and can probably be closed if you like.

closed 2017-06-06 by tpaviot

brunozilli, 2017-07-27:
Hello Thomas,

Please just read these few comments and give me a your idea.

I've tried to install python-occ with conda python 2.7 and it seems that all was fine. No error message during installation.
After this, I compiled pygear.py by python 2.7 and it worked with non error messages.

Then, I launched pygear but it returned me  a message:

caelinux@Bruno:~/Documenti/pygear-0.24$ python /usr/local/lib/python2.7/dist-packages/pygear.pyc
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/pygear.py", line 37, in <module>
    from OCC.TColgp import TColgp_Array1OfPnt2d, TColgp_Array1OfPnt

Thank you very much for your help and ideas.

Bruno