tpaviot/pythonocc-core #393: Low default quality rendering for high curvature surfaces in Qt

p-chambers, 2017-01-23:
I'm having some issues with the rendering of surfaces with a high curvature (wing shapes, to be precise): See p-chambers/occ_airconics#18 for the original issue, image included below.

[image]

I don't think it's an issue with the shape itself, as I exported it as .step and loaded it into Rhino, which produced the smooth rendering shown on the [occ_airconics readthedocs](http://occ-airconics.readthedocs.io/en/latest/).

There doesn't seem to be an obvious way to increase the Qt viewer mesh quality: can it be done?

jf---, 2017-01-23:
hi @p-chambers,

this is a way of increasing the allowed deviation

```
display.Context.SetDeviationAngle(display.Context.DeviationAngle() / 100.)
display.Context.SetDeviationCoefficient(display.Context.DeviationCoefficient() / 100.)
``

more elegant is perhaps to set the defaults through `display.Context.DefaultDrawer().GetObject()`

p-chambers, 2017-01-23:
Thanks Jelle, this solves the issue for me. I had experimented with setting the deviation coefficient only and it didn't seem to help, but refining both the coefficient and the angle as you suggest makes a more reassuring rendering of the shape.

Oddly, Mac Coacoa window doesn't seem to have as much of a problem with rendering at the default values, so this may be graphic driver dependent to some extent. However, I'm happy that this has solved the issue.

closed 2017-01-23 by p-chambers