tpaviot/pythonocc-core #398: I have Writing python code to find Minimum bounding box,but issue is that we can't got a correct bounding box if shap is cylindrical

chandrabhangholap, 2017-02-03:


chandrabhangholap, 2017-02-03:
my code is 
topology_=Topo(stepShape)
x_Min= sys.float_info.max
y_Min = sys.float_info.max
z_Min = sys.float_info.max
x_Max= -sys.float_info.max
y_Max = -sys.float_info.max
z_Max = -sys.float_info.max

for vrtx in topology_.vertices():
    x_l = BRep_Tool.Pnt(vrtx).X()
    y_l = BRep_Tool.Pnt(vrtx).Y()
    z_l = BRep_Tool.Pnt(vrtx).Z()

    if x_l<x_Min:
        x_Min = x_l
    if y_l<y_Min:
        y_Min = y_l
    if z_l < z_Min:
        z_Min=z_l
    if x_l>x_Max:
        x_Max = x_l
    if y_l>y_Max:
        y_Max = y_l
    if z_l>z_Max:
        z_Max=z_l

print(x_Min)
print(y_Min)
print(z_Min)
print(x_Max)
print(y_Max)
print(z_Max)
print(x_Max - x_Min)
print(y_Max - y_Min)
print(z_Max - z_Min)

tpaviot, 2017-04-02:
@chandrabhangholap can you please be a bit more precise about what is expected and which shape is passed to your code.

tpaviot, 2017-05-04:
@chandrabhangholap closing issue, not enough inputs to reproduce/identify/fix the bug

closed 2017-05-04 by tpaviot