tpaviot/pythonocc-core #400: Using Qt5: "Cannot create platform OpenGL context ... " error, solved using Qt4

calocedrus, 2017-02-08:
On a linux box (ubuntu 14.04), 
`conda create -n pythonocc -c pythonocc -c dlr-sc pythonocc-core==0.17.2 python=3`
installs qt5 and the corresponding pyqt5. 
Then when trying to run an example such as 
`py core_load_iges.py`
I get the following: 

```
Total number of loaded entities : 4.
INFO:OCC.Display.backend:backend loaded: qt-pyqt5
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt5
**QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled**
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
Xw_Window created.
Display3d class successfully initialized
```

The display window opens correctly, but the actual object occupies a very tiny area at the top left corner of the display window.
To solve this, I have first removed pythonocc and qt5, then installed qt4 followed by pythonocc. The above example then opens correctly, with the actual object occupying the full display window as expected.
The issue is 
`QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled`
and seems to be related to:
https://github.com/ContinuumIO/anaconda-issues/issues/1267
(so likely not a pythonocc issue but I thought I'd still report it here).

jf---, 2017-02-09:
@calocedrus , why not [try another pyqt5](https://anaconda.org/search?q=pyqt5), that might help you rule out the problem

calocedrus, 2017-02-13:
@jf  I thought I had posted a reply last week but I must have closed my browser without actually posting my comment.
Thanks for the idea, following your comment I did try many other pyqt5. 
For example:
`conda install -c astrofrog pyqt5`
But it is to no avail:
Installing a pyqt5 will still have pythonocc-core to install its own pyqt:
```
conda install -c pythonocc -c dlr-sc pythonocc-core==0.17.2 python=3
The following packages will be REMOVED:
    pyqt:           5.6.0-py35_2           
    pythonocc-core: 0.17.2-py35_1 pythonocc

```
I can't find out how to remove this pythonocc installed pyqt without also removing pythonocc.
Note that pythonocc-core installs a version 5 of pyqt (pyqt--- and not pyqt5---), but from searching for pyqt packages on anaconda, it seems that all but anaconda's own pyqt packages relate to a version 4; I'm not too sure about the naming conventions here (pyqt is for version 4 or 5?). How to have pythonocc to be satisfied with an alternative pyqt5 rather than its own (or anaconda's) pyqt?

jf---, 2017-02-14:
@calocedrus i'd uninstall the old pyqt using conda and try to install pyqt5 from an alternative channel. also there's tools in conda that'll inform you on the packages installed in a certain environment

calocedrus, 2017-02-15:
@jf I couldn't find how to uninstall pyqt without it automatically also uninstalling pythonocc, and `--force` is what was needed:
`conda uninstall --force qt pyqt`
Now it seems installing an alternative pyqt5 will also install pyqt (it is the case for the several alternatives I've tried):
```
$ conda install -c astrofrog pyqt5
Fetching package metadata ...........
Solving package specifications: .
Package plan for installation in environment /home/ludovic/anaconda3/envs/test:
The following NEW packages will be INSTALLED:
    pyqt:  5.6.0-py35_2          
    pyqt5: 5.5.0-py35_0 astrofrog
    qt:    5.6.2-3               
    qt5:   5.5.0-0      astrofrog
Proceed ([y]/n)?
```
I still proceeded and then re-uninstalled pyqt and qt. So now, in my pythonocc dedicated test environment, `conda list` shows me only qt5 and pyqt5, no other qt nor pyqt.
But then when testing an example, pythonocc told me I had no compliant gui available:
```
$ py core_load_iges.py 
Total number of loaded entities : 4.
Traceback (most recent call last):
  File "core_load_iges.py", line 38, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "~/anaconda3/envs/test/lib/python3.5/site-packages/OCC/Display/SimpleGui.py", line 92, in init_display
    used_backend = load_backend(backend_str)
  File "~/anaconda3/envs/test/lib/python3.5/site-packages/OCC/Display/backend.py", line 214, in load_backend
    raise ImportError("No compliant GUI library could be imported.\n"
ImportError: No compliant GUI library could be imported.
Either PyQt5, PyQt4, PySide, or wxPython is required
```
It seems pythonocc expect specifically a pyqt and not a pyqt5 package... For now I give up trying to solve this since I have pythonocc working fine with pyqt4. Thank you for your guidance so far.

jf, 2017-02-15:
@calocedrus I don't know if @jf--- has told you already... but "@ jf" is a different person. Get your references right, pls

calocedrus, 2017-02-15:
Had no idea! I've updated all references. Sorry, jf.

tpaviot, 2017-05-04:
The linux issue is solved by commit d90e0c3

Please keep using pyqt5.

closed 2017-05-04 by tpaviot

AutoIC, 2018-09-11:
@calocedrus 
I had the same issue when using pythonocc with pyqt5 on windows, python 3.5. 

The main error message was saying "ImportError: No compliant GUI library could be imported." when I tried to run the very first example "core_2d_fillet.py"

It turned out this workaround solved it for me: 
pip install pyqt5==5.7 sip==4.18

https://github.com/ContinuumIO/anaconda-issues/issues/1267

You may also want to make sure that your CASROOT and CSF_ShadersDirectory environment variables are set correctly, in my case I set them to \oce-0.18.3\share\oce and \oce-0.18.3\share\oce\src\Shaders Respectively.