tpaviot/pythonocc-core #402: Cannot DownCast from AIS_InteractiveObject

smirnovskysaha, 2017-02-16:
I want to get a selected object which was displayed through AIS_Shape. I run this code:
`
import OCC
from OCC.Display.SimpleGui import init_display
from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox

display, start_display, add_menu, add_function_to_menu = init_display()

box = BRepPrimAPI_MakeBox(100,100,100).Shape()
anAis = OCC.AIS.AIS_Shape(box)

ais_context = display.GetContext().GetObject()
ais_context.SetCurrentObject(anAis.GetHandle())
ais_context.InitCurrent()
currentAis = ais_context.Current()

shape = OCC.AIS.Handle_AIS_Shape_DownCast(currentAis)
`

and I get 
TypeError: in method 'Handle_AIS_Shape_DownCast', argument 1 of type 'Handle_Standard_Transient const &'

What do I want is to get
shape == box is True

smirnovskysaha, 2017-02-16:
I got it. I have created my own Python wrapper with SWIG for simple function:
```
Handle(AIS_Shape) myDownCast(Handle(AIS_InteractiveObject) anIO)
{
    return Handle(AIS_Shape)::DownCast(anIO);
}
```
And it works well. So there is a bug with Handle_AIS in PythonOCC.

jf---, 2017-02-20:
Hi @smirnovskysaha ,

Interesting how you bypassed the issue. 
That said, the `AIS_Shape` has a `.Shape` method which refers to the `TopoDS_Shape`

so I would expect `currentAis.Shape() == box ` ...

smirnovskysaha, 2017-02-20:
currentAIS is the Handle object, currentAIS.GetObject() returns class<'AIS_InteractiveObject'>. To get AIS_Shape one should downcast from currentAIS to AIS_Shape. I found this method here: https://www.opencascade.com/content/how-convert-aisinteractiveobject-aisshape

jf---, 2017-02-20:
Ah, and `AIS_InteractiveObject` lacks the `.Shape` method...
Understood, thanks for your feedback...

Its true that some limitations apply when delving deeper into `AIS`, such as subclassing from `AIS_Shape`. So I think your idea of extending the wrapper here is a valuable idea...

tpaviot, 2017-05-03:
@smirnovskysaha in pythonocc, the class hierarchy is broken, and the Handle_AIS_Shape class does not inherit from Handle_Standard_Transient. Thus the error. Thanks for reporting this issue, don't know how to fix it so far, I'll let you know

smirnovskysaha, 2017-05-03:
Thomas, thank you for the feedback!

tpaviot, 2017-05-03:
Fixed by PR #418

tpaviot, 2017-05-03:
ping @smirnovskysaha

closed 2017-05-03 by tpaviot

smirnovskysaha, 2017-05-04:
Thank you, Thomas! I'll try new version later and write if there is OK.

smirnovskysaha, 2017-05-10:
I have tried new version and it works fine! Thanks once more!