tpaviot/pythonocc-core #403: SMESH and netgen support

tpaviot, 2017-02-20:
@trelau FYI I mangaed to get pythonocc/smesh672/netg62 compiled.

uploaded to https://anaconda.org/tpaviot/pythonocc-core

in order to test, you'll have to move smesh libraries from lib to bin in you env directory.  Please report any feedback

trelau, 2017-02-20:
@tpaviot great work. when i install from conda i get this error when trying to run the basic example:

[image]

I downloaded the package and manually copied and pasted the OCC package into the existing without replacing existing files so it only copied new SMESH items. I was able to run the basic example:

[image]

I've been toying around with the netgen mesher but it keeps fatally crashing. although it's too early to tell if i'm just not setting up the algos/hypothesis correctly.

[image]

I'm using a max length 1-d algo/hypothesis but the max size parameter at bottom is much larger than it should be. I've tried setting max size in a netgen_hypothesis_2d but no change. i'll keep experimenting.

That message seems to be printed here : https://github.com/tpaviot/smesh/blob/tp/netgen-support/src/NETGENPlugin/NETGENPlugin_Mesher.cpp#L531

tpaviot, 2017-02-21:
@trelau you have to uninstall the dlr-sc oce and install the official one:

``` conda remove oce
conda install -c oce=0.17.2
```

indeed dlr-sc applies a patch over the official oce that makes it non binary compliant.

trelau, 2017-02-21:
@tpaviot resolved. thanks.

I'm following the netgen process shown in FreeCAD [here](https://github.com/FreeCAD/FreeCAD/blob/3377b4264220b938c27aa7c4e5aab3f710173bee/src/Mod/MeshPart/App/Mesher.cpp#L312). If I do that I get a message saying a sub mesh is not computed on an edge. Before, I applied a 1D hypothesis/algorithm and had to set anUpward=True in aMeshGen.Compute() but saw the error posted above.

[image]

Here is the method added to the smesh example 

```
def smesh_netgen(event=None):
    # Create the Mesh
    aMeshGen = SMESH_Gen()
    aMesh = aMeshGen.CreateMesh(0, True)
    aMesh.ShapeToMesh(aShape)

    # 2D
    hypo2d = NETGENPlugin_Hypothesis_2D(0, 0, aMeshGen)
    hypo2d.SetFineness(2)
    hypo2d.SetQuadAllowed(True)
    hypo2d.SetOptimize(True)
    hypo2d.SetSecondOrder(False)
    algo2d = NETGENPlugin_NETGEN_2D(1, 0, aMeshGen)

    aMesh.AddHypothesis(aShape, 0)
    aMesh.AddHypothesis(aShape, 1)

    # Compute the data
    aMeshGen.Compute(aMesh, aMesh.GetShapeToMesh())

    # Display the data
    display_mesh(aMesh)
```

tpaviot, 2017-02-21:
@trelau I think you have to use the NETGENPlugin_Mesher class, Compute() method and SetParameters to add hypothesis. The NETGENPlugin_Mesher class was added a few minutes ago to the wrapper

tpaviot, 2017-02-21:
@trelau forget my previous comment, it's an issue related to the c++ code in NETGENPlugin_Mesher.cpp

jf---, 2017-02-21:
Very exciting work with the NETGEN plugin, <wow>!

trelau, 2017-02-21:
@tpaviot so i attempted to build pythonocc w/ smesh 771 in case it resolved any of my netgen problems and although i got it built, netgen still crashes when you try and compute the mesh. i tried it with NETGENPlugin_Mesher.Compute() and the regular SMESH_Gen.Compute(). I have all the changes i made to the wrapper generator, netgen, and smesh so i can push them up later. it's not outputting any diagnostic messages so maybe i need to define a switch when i build smesh and/or netgen. here is a pic from the new medial axis projection algorithm (it looks like a quadrangle for now):

[image]

trelau, 2017-02-22:
@tpaviot @jf---  I tried to add a netgen test to smesh to figure out where the error is at. See commit [here](https://github.com/trelau/smesh/commit/eb36c4f9d1a30610acae81ca8d4eab71ab18a78c). It fails at NETGENPlugin_Mesher.Compute(). I tried to follow the FreeCAD code [here](https://github.com/FreeCAD/FreeCAD/blob/d0dc4d10534ad0af29707cd72f55967ba58738cc/src/Mod/Fem/App/FemMeshShapeNetgenObject.cpp#L84). Hopefully this provides some clues. It seems like smesh/netgen connection is flawed...

tpaviot, 2017-02-22:
@trelau I'm not sure netgen-6.2 occ/oce support is ok, there's a warning about that when you select oce in cmake. Try with netgen6.1, I updated the smesh code see https://github.com/tpaviot/smesh/tree/tp/netgen61-support. you have to checkout the 6.1 branch from the sf git repository. I did not test it myself, I just got it compiled on linux

trelau, 2017-02-22:
Hmmm...before i started working on the smesh integration, I remember being able to build the netgen application (gui) with OCE 0.17, if that means anything. I'll try 6.1 when I get a chance.

trelau, 2017-02-22:
@tpaviot @jf---  so after some digging and thanks to @looooo , I am able to successfully run the netgen plugin with my smesh 771 version. see the commits [here](https://github.com/looooo/netgen/commit/d0f3903d1599f9cd60b1dba6f27ea0bc6a258507) and [here](https://github.com/trelau/netgen/commits/master) for what i think is the fix.

[image]

i'll get some things cleaned up and try and get my smesh 771 branch pushed up for future reference. my guess is this should fix your smesh 671 w/ netgen 6.2.

trelau, 2017-02-22:
@tpaviot for reference, i pushed up my latest netgen 6.2, smesh 771, pythonocc-core, and python-generator (both for smesh 771). I also have netgen and smesh on conda for win32/64. I was able to locally build pythonocc using the installed conda netgen and smesh apps.

Which ever way you decide to proceed (smesh 6.7 or migrate to 7.7.1) I'll start experimenting with it and put together docs and examples.

trelau, 2017-02-23:
@tpaviot I found a test case that failed for netgen when a face has internal edges like shown below

[image]

smesh throws some errors and you're left with this...

[image]

i manually applied some of the netgen 53 patches for Salome and it appears to resolve the issue

[image]

i'll finish applying the patches (other look to be optimizations) in [this branch](https://github.com/trelau/netgen/tree/netgen62ForSMESHPatch).

When it comes time, let me know how you'd like to proceed. I assume you'll go back to your original netgen 6.2 process (cloning source and applying patches). We can make this a netgen62ForSMESH.patch or something.

tpaviot, 2017-02-23:
@trelau your experiments are convincing, they are a proof of concept. I spent many hours testing different configurations involving smesh671/sesh771/netgen61/netgen62/linux/windows. Here is what I suggest to do in the coming days:
* first of all, I'd like to release what is already achieved: smesh671 support in pythonocc-0.17.3-dev, without netgen support. I plan to release at the end of the next week (need a bit more polishing, testing, conda deployment on all platforms and a couple of days of vacation in the meantime). This will be the pythonocc-0.17.3 release ;

* after that, we will move towards smesh771/netgen62 integration into pythonocc-0.17.4

* I suggest you're the maintener of the patched netgen62. I will remove all my netgen experiments/repositories, there's no need that we both do the same thing, and that's something that takes time. IMO the best way to proceed is that you create a branch netgen4mesh, or something like that, on top of your netgen fork. You include in that branch all the patches you applied over the official netgen repository. Then you regularly update the master branch, and rebase on top of master to reflect upstream changes to your netgen4smesh branch. You can notify the FreeCad team, they might be interested, as well as the netgen maintener (FYI, there's a freecad patch available on the sf.net netgen project page, which could be included) ;

* on my side, I take care of updating smesh771 to support your patched netgen, and upgrade the python wrapper.

Note that my concern is to have all of that working on both linux/osx and windows platforms.

trelau, 2017-02-23:
@tpaviot that sounds like a good plan.

If I understand correctly, you are suggesting I keep my current Netgen fork and add another branch (with the source) of the patches? My initial thought was I'd set up a repo similar to your netgen-conda, where the source is downloaded during build and patches applied. I have little experience with these types of workflows so i just want to make sure i go down the most efficient path and understand your suggestion correctly. Thanks again for your support.

tpaviot, 2017-02-23:
@trelau I think it's better that you keep going on with your repository. Just move your commits to a separate branch so that it's easy to check your changes (you can also clean up the history with git rebase). My netgen-conda (i.e. dowload + a set of patches) is perfect is you only have a very small set of patches, let's say three or four. But it's not convenient if the number of patches is higher and/or if you want to manage contributions from other people (for example if I  have a patch related to linux/osx, or the freecad guys, or anyone else).

trelau, 2017-02-23:
@tpaviot ok i will proceed that way. thanks.

rainman110, 2017-06-29:
@tpaviot Here's some feedback to SMESH on macOS.

This post is a duplicate to: https://github.com/tpaviot/smesh/issues/45

The smesh conda packages cannot be loaded by python. The _SMESH package cannot be imported, as the libraries are not properly fixed up.

The error are similar to:
```
Traceback (most recent call last):
  File "/Users/martin/Documents/miniconda/envs/pythonocc-nightly/lib/python3.6/site-packages/OCC/SMESH.py", line 18, in swig_import_helper
    return importlib.import_module(mname)
  File "/Users/martin/Documents/miniconda/envs/pythonocc-nightly/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 978, in _gcd_import
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 648, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 560, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 922, in create_module
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
ImportError: dlopen(/Users/martin/Documents/miniconda/envs/pythonocc-nightly/lib/python3.6/site-packages/OCC/_SMESH.so, 2): Library not loaded: libMEFISTO2.3.dylib
  Referenced from: /Users/martin/Documents/miniconda/envs/pythonocc-nightly/lib/libStdMeshers.3.0.0.dylib
```
You should enable binary relocation or use install_name_tool to fix these problems.

tpaviot, 2017-06-30:
@rainman110 ok I'll do this. I don't have any osx machine, I only use the travis build to compile/check/distribute. I'll rebuild smesh with binary relocation, and notify you so that you can test if it works.

jf---, 2017-06-30:
iirc, relocatable binaries were costly to built ( in terms of time ), since re-writing the relative paths to the generated binaries took *forever*. so possible, to save limited Travis CI time, this was switched off. Possible a more effective strategy was chosen in more recent conda versions. This was the state of things about a year ago or so.

rainman110, 2017-06-30:
@jf If compile time is the issue, then one should correct the paths manually.

The first step would be, to enable RPATHS on macOS as it is also done in OCE. This would solve part of the problem. Then, one should use install_name_tool to hand-tweak the remaining issues.

tpaviot, 2017-07-01:
@rainman110 do you mean enabling binary relocation in the pythonocc-core receipe or smesh conda receipe ? Please contribute any patch that would solve osx issues. Since I don't have any osx machine, but only travis reports, it's a kind of "open loop" process for me, I can't get any additional feedback.

If this is a pythonocc-core related discussion, let's go on here. If it's rather something related to smesh, I suggest we have this discussion on the smesh issue tracker.

rainman110, 2017-07-01:
@tpaviot With the newest SMESH release (https://github.com/tpaviot/smesh/commit/672518002ae650ff6666eb5e3c93d5a2f3ab9bc6), it works on macOS when I compile smesh and pythonocc by my own. The RPATH stuff did the trick.

I'll try to build recent conda packages and give feedback again when I am done.

matpen, 2018-10-12:
Sorry to ping, but are there any news about this?

I was trying to use [PYTHONOCC_WRAP_SMESH_NETGENPLUGIN](https://github.com/tpaviot/pythonocc-core/blob/5d5038312f0fddd1c64a63f4fbdb59eaab37d708/CMakeLists.txt#L145) while building pythonOCC, then realized that [SMESH_BUILD_NETGENPLUGIN](https://github.com/tpaviot/smesh/blob/a6780b8476bc6ce022275c07d4814abd8254605f/CMakeLists.txt#L28) is currently disabled. 

This led me to [this](https://github.com/tpaviot/smesh/issues/8#issuecomment-278883448) discussion, which seems to imply that the support for netgen is now possible. Also found [this repo](https://github.com/mixxen/smesh-netgen/graphs/contributors), which however has not got any activity recently.

tpaviot, 2020-01-05:
Deprecated. SMESH is not targetted anymore by the pythonocc wrapper.

closed 2020-01-05 by tpaviot