tpaviot/pythonocc-core #404: Wrapping SMDS_ElemIteratorPtr and SMDS_NodeIteratorPtr

trelau, 2017-03-01:
SMDS_ElemIteratorPtr and SMDS_NodeIteratorPtr appear the way you access and iterate through nodes and elements within a mesh. When I try and access a method that returns one of these types, I get the following message:

[image]

I tried a few things based on the info [here ](http://swig.org/Doc3.0/Library.html#Library_std_shared_ptr) and [here](http://stackoverflow.com/questions/18878179/how-to-avoid-memory-leak-with-shared-ptr-and-swig), which at this point is simply adding this to the SMDS.i file:

```
%include "boost_shared_ptr.i"
%shared_ptr(SMDS_ElemIteratorPtr)
%shared_ptr(SMDS_NodeIteratorPtr)
```

and got the same results. I'll keep working it, but it seems like a necessary functionality to be able to work with meshes.

tpaviot, 2017-03-11:
@trelau seems to work, thanks see 84d1968a4677adaf86f6ab401cde5f50a979542c

tpaviot, 2017-06-06:
Fixed by commit 5ca2062c9a127f6ab74f2629e0819798b8f0d1d0

closed 2017-06-06 by tpaviot