tpaviot/pythonocc-core #405: Anaconda (python3.6 ver) failed installing pythonocc

TsingQAQ, 2017-03-04:
Hi there, I'm using the latest ANACONDA 4.3 to install pythonocc using command
`conda install -c dlr-sc pythonocc-core=0.17`
But it returns:

`_UnsatisfiableError: The following specifications were found to be in conflict:`
` python 3.6*`
`pythonocc-core 0.17* -> python 2.7*`
`Use "conda info <package>" to see the dependencies for each package._`

I'm not willing to use conda create env instruction cause I'd like to install it directly in anaconda's Lib to make use of some pre-exist packages,  though the installation seems failed at the moment. So I wonder how to fix this problem, is that the reason pythonocc support python3.5 so a older versions of Anaconda is needed? 
thanks in advance.

tpaviot, 2017-03-04:
@TsingQAQ conda moved up to python3.6 after pythonocc-0.17.2 was released.  The coming 0.17.3 version will support python3.6 as well

tpaviot, 2017-03-08:
Fixed by commit 242ad5f2e04353864880b1681163234def51216b

closed 2017-03-08 by tpaviot

thorade, 2017-05-19:
For me, the follwoing command still fails:
`conda install -c DLR-SC pythonocc-core>=0.17.3`
with error message
```
UnsatisfiableError: The following specifications were found to be in conflict:
  - python 3.6*
  - pythonocc-core -> python 2.7*
Use "conda info <package>" to see the dependencies for each package.
```

Am I doing something wrong?
This is the first time ever I use Anaconda and conda.

thorade, 2017-05-19:
I added three additional channels (in this order), then all packages are found somewhere:
```
conda config --append channels DLR-SC
conda config --append channels pythonocc
conda config --append channels oce
conda install pythonocc-core
```