tpaviot/pythonocc-core #408: Viewer Freeze

LukasMHD, 2017-03-15:
Hello,
I have a problem, I have set up ifcopenshell and pythonocc on a new System (Laptop, Windows 10, Python 2.7, Miniconda(for 2.7), pythonocc 0.17.2, everything 64bit).
but whenever i initialize the viewer via an IDE 

`import ifcopenshell, ifcopenshell.geom`
`display = ifcopenshell.geom.utils.initialize_display()`

the display window starts, but imidiately freezes as well. The Header just says
'pythonOCC-0.17.2 3d viewer ('qt-pyqt5' backend) (Not Responding)'
and thats it.
I got it running on my PC, and I cannot, for the life of me, figure out why it shoudn't work on my Laptop as well.
Thanks for any suggestions.
king regards,
Luaks.

adam-urbanczyk, 2017-04-18:
Just a guess: sounds like the pyqt event loop is blocked / not running.

tpaviot, 2017-04-19:
@LukasMHD I suggest you post to the ifcopenshell issue tracker at https://github.com/IfcOpenShell/IfcOpenShell/issues @aothms

closed 2017-04-19 by tpaviot

aothms, 2017-04-19:
I also have bad experiences running the pythonocc viewer from some IDEs. Somehow they do seem to interfere with the event loop. Haven't had the time to look into it. Judging from the lack of issues I find online, I'm inclined to say it does not seem to be a generic issue with PyQt and the IDE.

tpaviot, 2017-09-25:
@Luccifer Closed because this issue seems to be related to ifcopenshell. If you manage to get a pythonocc only issue, please submit a new issue

Luccifer, 2017-09-25:
All right then! Thanks for participating!

jf---, 2017-10-02:
@aothms , @adam-urbanczyk  is right here, this is due to the event loop. ipython can much facilitate developing complex pyocc programs, by having a "primed" eventloop ( belonging to ipython, rather than pyocc, which then uses the singleton ). This works well with dynamic reloading; you dont have to restart the program from scratch, but can develop the interactive app, "interactively".

aothms, 2017-10-06:
@jf--- that sounds like a killer feature, but I don't think its the kind of interactivity that is necessary in this simple case. If @Luccifer or @LukasMHD want to spend the effort, it wouldn't be difficult to create see whether this is an IfcOpenShell or pythonOCC related issue. `ifcopenshell.geom.utils.initialize_display()` is a pretty thin wrapper around `OCC.Display.SimpleGui.init_display()`. I don't have an IDE installed, but I have heard others complain.

https://github.com/IfcOpenShell/IfcOpenShell/blob/master/src/ifcopenshell-python/ifcopenshell/geom/occ_utils.py#L48

Luccifer, 2017-10-06:
Okey, I am little bit tiered of notifications, but I fixed the problem fast with 
`ifcopenshell.geom.utils.main_loop()`
Never worked with Threads/Pipes with python, only compiled languages with UI itself