tpaviot/pythonocc-core #410: Salome GEOMAlgo_Splitter concept available

trelau, 2017-04-01:
Just wanted to make this available in case you want to incorporate it into future releases...

The Salome GEOMAlgo_Splitter was always a useful tool in Salome so I migrated it to pythonocc as shown by [this commit](https://github.com/trelau/pythonocc-core/commit/30d71bc424aaacb5a39e50bba675fff0a37e8850).

The GEOMAlgo_Splitter header and source files are from Salome 8.2.0 release. The example below shows splitting two faces and an edge. You can use it by importing:

`from OCC.GEOMAlgo import GEOMAlgo_Splitter`

[image]

[image]

tpaviot, 2017-04-01:
@trelau interesting. Can you please create a PR ? (if you can add an example as well it would be cool)

tpaviot, 2017-04-02:
Related PR is #411

tpaviot, 2017-04-02:
Merged into master.

closed 2017-04-02 by tpaviot