tpaviot/pythonocc-core #413: Iussues with pythonocc-core 0.17.3 and Anaconda 4.3.1 (Python 3.6) on Windows

mahibra, 2017-04-15:
I am currently trying to switch to new versions of Anaconda and pythonocc-core on my Windows PC.

First I installed Anaconda 4.3.1. Then I tried to installing pythonocc-core using 
`conda install -c pythonocc -c oce pythonocc-core==0.17.3`, which gave me the following errors:
```
PackageNotFoundError: Dependencies missing in current win-64 channels:
  - pythonocc-core ==0.17.3 -> oce ==0.17.3 -> gl2ps
  - pythonocc-core ==0.17.3 -> oce ==0.17.3 -> tbb
```
I thus installed the two packages from the dlr-sc package and was then able to successfully install pythonocc-core.

When I try to run one of the examples, I get the following error:
```
c:\1\examples>python core_animation.py
INFO:OCC.Display.backend:backend loaded: qt-pyqt5
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt5
Traceback (most recent call last):
  File "core_animation.py", line 30, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "C:\Anaconda3\lib\site-packages\OCC\Display\SimpleGui.py", line 145, in init_display
    from OCC.Display.qtDisplay import qtViewer3d
  File "C:\Anaconda3\lib\site-packages\OCC\Display\qtDisplay.py", line 29, in <module>
    from OCC.Display import OCCViewer
  File "C:\Anaconda3\lib\site-packages\OCC\Display\OCCViewer.py", line 27, in <module>
    from OCC.AIS import AIS_MultipleConnectedInteractive, AIS_Shape
  File "C:\Anaconda3\lib\site-packages\OCC\AIS.py", line 22, in <module>
    _AIS = swig_import_helper()
  File "C:\Anaconda3\lib\site-packages\OCC\AIS.py", line 21, in swig_import_helper
    return importlib.import_module(mname)
  File "C:\Anaconda3\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: DLL load failed: Das angegebene Modul wurde nicht gefunden.
```

tpaviot, 2017-04-16:
@mahibra Please try also ti install FreeImage from dlr-sc

mahibra, 2017-04-16:
I now started from a new Anaconda 4.3.1 installation and ran

```
c:\1\examples>conda install -c dlr-sc -c pythonocc -c oce pythonocc-core==0.17.3
Fetching package metadata .................
Solving package specifications: .

Package plan for installation in environment C:\Anaconda3:

The following NEW packages will be INSTALLED:

    freeimage:      3.17.0-vc14_3 dlr-sc    [vc14]
    gl2ps:          1.3.8-vc14_3  dlr-sc    [vc14]
    oce:            0.17.3-vc14_1 oce       [vc14]
    pythonocc-core: 0.17.3-py36_0 pythonocc
    tbb:            4.3.6-vc14_1  dlr-sc    [vc14]

The following packages will be UPDATED:

    conda:          4.3.14-py36_1           --> 4.3.16-py36_0
```

However, I still get the same error when executing a pythonocc sample file.

tpaviot, 2017-04-16:
There might be a conflict somewhere, I can't reproduce the issue, it works on my machine. Can you please run  ```conda list``` , ouput for my windows machine (32 bit):
```
(pyocc36) D:\Devel\pythonocc-core\examples>conda list
# packages in environment at D:\Miniconda2\envs\pyocc36:
#
freeimage                 3.17.0                   vc14_3  [vc14]  dlr-sc
freetype                  2.6.3                    vc14_1  [vc14]  dlr-sc
gl2ps                     1.3.8                    vc14_3  [vc14]  dlr-sc
icu                       57.1                     vc14_0  [vc14]
jpeg                      9b                       vc14_0  [vc14]  dlr-sc
libpng                    1.6.27                   vc14_0  [vc14]
oce                       0.17.3                   vc14_1  [vc14]  oce
openssl                   1.0.2k                   vc14_0  [vc14]
pip                       9.0.1                    py36_1
pyqt                      5.6.0                    py36_2
python                    3.6.1                         0
pythonocc-core            0.17.3                   py36_0    pythonocc
qt                        5.6.2                    vc14_3  [vc14]
setuptools                27.2.0                   py36_1
sip                       4.18                     py36_0
tbb                       4.3.6                    vc14_1  [vc14]  dlr-sc
vs2015_runtime            14.0.25123                    0
wheel                     0.29.0                   py36_0
zlib                      1.2.8                    vc14_3  [vc14]
```

mahibra, 2017-04-16:
Looks like you are on the right path. I just installed Miniconda and am able to run pythonocc. Here are the packages included in the miniconda installation:
```
(C:\Miniconda3) c:\1\examples>conda list
# packages in environment at C:\Miniconda3:
#
cffi                      1.9.1                    py36_0
conda                     4.3.16                   py36_0
conda-env                 2.6.0                         0
console_shortcut          0.1.1                    py36_1
cryptography              1.7.1                    py36_0
freeimage                 3.17.0                   vc14_3  [vc14]  dlr-sc
freetype                  2.6.3                    vc14_1  [vc14]  dlr-sc
gl2ps                     1.3.8                    vc14_3  [vc14]  dlr-sc
icu                       57.1                     vc14_0  [vc14]
idna                      2.2                      py36_0
jpeg                      9b                       vc14_0  [vc14]  dlr-sc
libpng                    1.6.27                   vc14_0  [vc14]
menuinst                  1.4.4                    py36_0
oce                       0.17.3                   vc14_1  [vc14]  oce
openssl                   1.0.2k                   vc14_0  [vc14]
pip                       9.0.1                    py36_1
pyasn1                    0.1.9                    py36_0
pycosat                   0.6.1                    py36_1
pycparser                 2.17                     py36_0
pyopenssl                 16.2.0                   py36_0
pyqt                      5.6.0                    py36_2
python                    3.6.0                         0
pythonocc-core            0.17.3                   py36_0    pythonocc
pywin32                   220                      py36_2
qt                        5.6.2                    vc14_3  [vc14]
requests                  2.12.4                   py36_0
ruamel_yaml               0.11.14                  py36_1
setuptools                27.2.0                   py36_1
sip                       4.18                     py36_0
six                       1.10.0                   py36_0
tbb                       4.3.6                    vc14_1  [vc14]  dlr-sc
vs2015_runtime            14.0.25123                    0
wheel                     0.29.0                   py36_0
zlib                      1.2.8                    vc14_3  [vc14]
```
And these are the packages of the full Anaconda installation:
```
_license                  1.1                      py36_1
alabaster                 0.7.9                    py36_0
anaconda                  4.3.1               np111py36_0
anaconda-client           1.6.0                    py36_0
anaconda-navigator        1.5.0                    py36_0
anaconda-project          0.4.1                    py36_0
astroid                   1.4.9                    py36_0
astropy                   1.3                 np111py36_0
babel                     2.3.4                    py36_0
backports                 1.0                      py36_0
beautifulsoup4            4.5.3                    py36_0
bitarray                  0.8.1                    py36_1
blaze                     0.10.1                   py36_0
bokeh                     0.12.4                   py36_0
boto                      2.45.0                   py36_0
bottleneck                1.2.0               np111py36_0
bzip2                     1.0.6                    vc14_3  [vc14]
cffi                      1.9.1                    py36_0
chardet                   2.3.0                    py36_0
chest                     0.2.3                    py36_0
click                     6.7                      py36_0
cloudpickle               0.2.2                    py36_0
clyent                    1.2.2                    py36_0
colorama                  0.3.7                    py36_0
comtypes                  1.1.2                    py36_0
conda                     4.3.16                   py36_0
conda-env                 2.6.0                         0
configobj                 5.0.6                    py36_0
[... 122 lines cut ...]
sockjs-tornado            1.0.3                    py36_0
sphinx                    1.5.1                    py36_0
spyder                    3.1.2                    py36_0
sqlalchemy                1.1.5                    py36_0
statsmodels               0.6.1               np111py36_1
sympy                     1.0                      py36_0
tbb                       4.3.6                    vc14_1  [vc14]  dlr-sc
tk                        8.5.18                   vc14_0  [vc14]
toolz                     0.8.2                    py36_0
tornado                   4.4.2                    py36_0
traitlets                 4.3.1                    py36_0
unicodecsv                0.14.1                   py36_0
vs2015_runtime            14.0.25123                    0
wcwidth                   0.1.7                    py36_0
werkzeug                  0.11.15                  py36_0
wheel                     0.29.0                   py36_0
widgetsnbextension        1.2.6                    py36_0
win_unicode_console       0.5                      py36_0
wrapt                     1.10.8                   py36_0
xlrd                      1.0.0                    py36_0
xlsxwriter                0.9.6                    py36_0
xlwings                   0.10.2                   py36_0
xlwt                      1.2.0                    py36_0
zlib                      1.2.8                    vc14_3  [vc14]
```

mahibra, 2017-04-18:
I just want to let you know that I was able to reproduce this issue on a different computer with Windows 7, 64 Bit.

tpaviot, 2017-04-19:
@mahibra can you please give another try ? I updated the oce-0.17.3 package so that it depends on freeimageplus

hinxx, 2017-04-19:
@tpaviot, I tried it on linux-64 and I'm seeing similar problem with install.

	$ conda create -n pythonocc -c pythonocc -c oce pythonocc-core==0.17.3 python=2
	Fetching package metadata .............
	Solving package specifications: 

	PackageNotFoundError: Dependencies missing in current linux-64 channels: 
	  - pythonocc-core ==0.17.3 -> oce ==0.17.3 -> freeimageplus
	  - pythonocc-core ==0.17.3 -> oce ==0.17.3 -> gl2ps
	  - pythonocc-core ==0.17.3 -> oce ==0.17.3 -> tbb
	  - pythonocc-core ==0.17.3 -> oce ==0.17.3 -> freeimageplus
	  - pythonocc-core ==0.17.3 -> oce ==0.17.3 -> gl2ps
	  - pythonocc-core ==0.17.3 -> oce ==0.17.3 -> tbb

	Close matches found; did you mean one of these?

		freeimageplus: freeimage
		tbb: tblib

	(and similarly for the other packages)

	$ conda list
	# packages in environment at /home/hinkokocevar/Tools/miniconda2:
	#
	anaconda-client           1.6.2                    py27_0  
	cffi                      1.9.1                    py27_0  
	clyent                    1.2.2                    py27_0  
	conda                     4.3.16                   py27_0  
	conda-env                 2.6.0                         0  
	cryptography              1.7.1                    py27_0  
	enum34                    1.1.6                    py27_0  
	freeimageplus             3.17.0                        0    DLR-SC
	idna                      2.2                      py27_0  
	ipaddress                 1.0.18                   py27_0  
	libffi                    3.2.1                         1  
	openssl                   1.0.2k                        0  
	pip                       9.0.1                    py27_1  
	pyasn1                    0.1.9                    py27_0  
	pycosat                   0.6.1                    py27_1  
	pycparser                 2.17                     py27_0  
	pyopenssl                 16.2.0                   py27_0  
	python                    2.7.13                        0  
	python-dateutil           2.6.0                    py27_0  
	pytz                      2017.2                   py27_0  
	pyyaml                    3.12                     py27_0  
	readline                  6.2                           2  
	requests                  2.12.4                   py27_0  
	ruamel_yaml               0.11.14                  py27_1  
	setuptools                27.2.0                   py27_0  
	six                       1.10.0                   py27_0  
	sqlite                    3.13.0                        0  
	tbb                       4.3.6                         0    DLR-SC
	tk                        8.5.18                        0  
	wheel                     0.29.0                   py27_0  
	yaml                      0.1.6                         0  
	zlib                      1.2.8                         3

hinxx, 2017-04-19:
weird, for some reason install worked after trying the command, instead:

`conda install -c oce -c pythonocc pythonocc-core=0.17.3`

I had to do `conda install -c DLR-SC gl2ps` prior to above command, though.

mahibra, 2017-04-21:
I reinstalled pythonocc-core 0.17.3 which now depends on
```
freeimageplus             3.17.0                   vc14_0  [vc14]  dlr-sc
```

Unfortunately, I still get the same error.

mahibra, 2017-04-21:
As mentioned above, pythonocc-core 0.17.3 works with Miniconda but not with Anaconda 4.3.1 (Windows 7, 64 Bit, Python 3.6.0)

2nicolas, 2017-04-30:
Same problem, when i launch a test:

> INFO:OCC.Display.backend:backend loaded: qt-pyqt5
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt5
Traceback (most recent call last):
  File "test.py", line 31, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "C:\ProgramData\Miniconda3\lib\site-packages\OCC\Display\SimpleGui.py", l
ine 145, in init_display
    from OCC.Display.qtDisplay import qtViewer3d
  File "C:\ProgramData\Miniconda3\lib\site-packages\OCC\Display\qtDisplay.py", l
ine 29, in <module>
    from OCC.Display import OCCViewer
  File "C:\ProgramData\Miniconda3\lib\site-packages\OCC\Display\OCCViewer.py", l
ine 27, in <module>
    from OCC.AIS import AIS_MultipleConnectedInteractive, AIS_Shape
  File "C:\ProgramData\Miniconda3\lib\site-packages\OCC\AIS.py", line 22, in <mo
dule>
    _AIS = swig_import_helper()
  File "C:\ProgramData\Miniconda3\lib\site-packages\OCC\AIS.py", line 21, in swi
g_import_helper
    return importlib.import_module(mname)
  File "C:\ProgramData\Miniconda3\lib\importlib\__init__.py", line 126, in impor
t_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: DLL load failed: Le module spécifié est [introuvable.]

After that, I check my _conda list_
> packages in environment at C:\ProgramData\Miniconda3:
cffi                      1.9.1                    py36_0
cmake                     3.6.3                    vc14_0  [vc14]
conda                     4.3.16                   py36_0
conda-env                 2.6.0                         0
console_shortcut          0.1.1                    py36_1
cryptography              1.7.1                    py36_0
freeimage                 3.17.0                   vc14_3  [vc14]  DLR-SC
freeimageplus             3.17.0                   vc14_0  [vc14]  dlr-sc
freetype                  2.5.5                    vc14_2  [vc14]
gl2ps                     1.3.8                    vc14_3  [vc14]  DLR-SC
icu                       57.1                     vc14_0  [vc14]
idna                      2.2                      py36_0
jpeg                      9b                       vc14_0  [vc14]  dlr-sc
libpng                    1.6.27                   vc14_0  [vc14]
menuinst                  1.4.4                    py36_0
oce                       0.17.3                   vc14_1  [vc14]  oce
openssl                   1.0.2k                   vc14_0  [vc14]
pip                       9.0.1                    py36_1
pyasn1                    0.1.9                    py36_0
pycosat                   0.6.1                    py36_1
pycparser                 2.17                     py36_0
pyopenssl                 16.2.0                   py36_0
pyqt                      5.6.0                    py36_2
python                    3.6.1                         0
pythonocc-core            0.17.3                   py36_0    pythonocc
pywin32                   220                      py36_2
qt                        5.6.2                    vc14_3  [vc14]
requests                  2.12.4                   py36_0
ruamel_yaml               0.11.14                  py36_1
setuptools                27.2.0                   py36_1
sip                       4.18                     py36_0
six                       1.10.0                   py36_0
swig                      3.0.10                        0
tbb                       4.3.6                    vc14_1  [vc14]  dlr-sc
vs2015_runtime            14.0.25123                    0
wheel                     0.29.0                   py36_0
zlib                      1.2.8                    vc14_3  [vc14]

I saw that the package _freetype_ doesn't come form dlr-sc.
So I update the package with
>conda install -c dlr-sc freetype
Fetching package metadata .............
Solving package specifications: .
Package plan for installation in environment C:\ProgramData\Miniconda3:
The following packages will be UPDATED:
    freetype: 2.5.5-vc14_2 [vc14] --> 2.6.3-vc14_1 dlr-sc [vc14]
Proceed ([y]/n)? y
freetype-2.6.3 100% |###############################| Time: 0:00:00 620.53 kB/s

Then...
>INFO:OCC.Display.backend:backend loaded: qt-pyqt5
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt5
QWindowsWindow::setGeometry: Unable to set geometry 1024x768+179+31 on QWidgetW
ndow/'MainWindowClassWindow'. Resulting geometry:  1024x749+179+31 (frame: 8, 3
, 8, 8, custom margin: 0, 0, 0, 0, minimum size: 0x0, maximum size: 16777215x16
77215).
 \###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
WNT window created.

Et cela marche.

tpaviot, 2017-05-02:
@2nicolas thanks for the tip, I'll try to force freetype dlr-sc 2.6.3

DEKHTIARJonathan, 2017-05-05:
hello @tpaviot,

I have the exact same issue, however I can not manage to install  freetype==2.6.3

I obtain the following error:
```
$ conda install -c dlr-sc freetype=2.6.3
Fetching package metadata ...............
Solving package specifications: .

UnsatisfiableError: The following specifications were found to be in conflict:
  - anaconda-navigator
  - freetype 2.6.3*
  - python 3.6*
Use "conda info <package>" to see the dependencies for each package.
```

When I list all the packages:
```
C:\WINDOWS\system32>conda list
# packages in environment at C:\ProgramData\Anaconda3:
#
_license                  1.1                      py36_1
alabaster                 0.7.9                    py36_0
anaconda                  4.3.1               np111py36_0
anaconda-client           1.6.0                    py36_0
anaconda-navigator        1.5.0                    py36_0
anaconda-project          0.4.1                    py36_0
astroid                   1.4.9                    py36_0
astropy                   1.3                 np111py36_0
babel                     2.3.4                    py36_0
backports                 1.0                      py36_0
beautifulsoup4            4.5.3                    py36_0
bitarray                  0.8.1                    py36_1
blaze                     0.10.1                   py36_0
bokeh                     0.12.4                   py36_0
boto                      2.45.0                   py36_0
bottleneck                1.2.0               np111py36_0
bzip2                     1.0.6                    vc14_3  [vc14]
cffi                      1.9.1                    py36_0
chardet                   2.3.0                    py36_0
chest                     0.2.3                    py36_0
click                     6.7                      py36_0
cloudpickle               0.2.2                    py36_0
clyent                    1.2.2                    py36_0
colorama                  0.3.7                    py36_0
comtypes                  1.1.2                    py36_0
conda                     4.3.17                   py36_0
conda-env                 2.6.0                         0
configobj                 5.0.6                    py36_0
console_shortcut          0.1.1                    py36_1
contextlib2               0.5.4                    py36_0
cryptography              1.7.1                    py36_0
curl                      7.52.1                   vc14_0  [vc14]
cycler                    0.10.0                   py36_0
cython                    0.25.2                   py36_0
cytoolz                   0.8.2                    py36_0
dask                      0.13.0                   py36_0
datashape                 0.5.4                    py36_0
decorator                 4.0.11                   py36_0
dill                      0.2.5                    py36_0
docutils                  0.13.1                   py36_0
entrypoints               0.2.2                    py36_0
et_xmlfile                1.0.1                    py36_0
fastcache                 1.0.2                    py36_1
flask                     0.12                     py36_0
flask-cors                3.0.2                    py36_0
freeimageplus             3.17.0                   vc14_0  [vc14]  dlr-sc
freetype                  2.5.5                    vc14_2  [vc14]
get_terminal_size         1.0.0                    py36_0
[... 104 lines cut ...]
spyder                    3.1.2                    py36_0
sqlalchemy                1.1.5                    py36_0
statsmodels               0.6.1               np111py36_1
sympy                     1.0                      py36_0
tbb                       4.3.6                    vc14_1  [vc14]  dlr-sc
tk                        8.5.18                   vc14_0  [vc14]
toolz                     0.8.2                    py36_0
tornado                   4.4.2                    py36_0
traitlets                 4.3.1                    py36_0
unicodecsv                0.14.1                   py36_0
virtualenv                15.1.0                    <pip>
vs2015_runtime            14.0.25123                    0
wcwidth                   0.1.7                    py36_0
werkzeug                  0.11.15                  py36_0
wheel                     0.29.0                   py36_0
widgetsnbextension        1.2.6                    py36_0
win_unicode_console       0.5                      py36_0
wrapt                     1.10.8                   py36_0
xgboost                   0.6a2                    py36_0    mikesilva
xlrd                      1.0.0                    py36_0
xlsxwriter                0.9.6                    py36_0
xlwings                   0.10.2                   py36_0
xlwt                      1.2.0                    py36_0
zlib                      1.2.8                    vc14_3  [vc14]
```

DEKHTIARJonathan, 2017-05-05:
Issue solved by running the following:
```
conda install -y -c conda-forge freetype=2.7
```

dfgordon, 2017-05-05:
I also had the freetype issue.  It appears if matplotlib is installed first, it seeds the environment with the wrong freetype, leading to headaches.  The following worked on Mac OS 10.12.4:
1) use miniconda
2) create a new, empty conda environment and switch to it
3) clean .bash_profile (e.g., undefine PYTHONPATH from MacPorts)
4) conda install -c oce -c pythonocc pythonocc-core
5) run "hello dumb box" example

dfgordon, 2017-05-05:
Correction:
4. conda install -c dlr-sc -c oce -c pythonocc pythonocc-core
I did it twice more, once on another Mac, and once on CentOS 7.3

mahibra, 2017-05-20:
I am happy to confirm that the following solved the issues with my full Anaconda 4.3.1 installation on Windows:
1. `conda install -c dlr-sc -c oce -c pythonocc pythonocc-core`
2. `conda install -y -c conda-forge freetype=2.7`

closed 2017-05-20 by mahibra