tpaviot/pythonocc-core #421: Export X3D and ThreeJS Down

DEKHTIARJonathan, 2017-05-05:
Hello,

I have been working with PythonOCC, and with python-occ 0.17.3 and 0.17.2 it is impossible to execute an export to X3D from a Tesselator Object :

**tess.ExportShapeToX3D(...)** gives the following error:
[image]

Concerning the export to ThreeJS, unfortunately the function "**ExportShapeToThreejs**" has been removed and the function to export to a JSON String gives nothing good:

```
{
	"metadata": {
		"version": 4.4,
		"type": "BufferGeometry",
		"generator": "pythonOCC"
	},
	"uuid": "data1/Actuator/part.step",
	"type": "BufferGeometry",
	"data": {
	"attributes": {
			"position": {
				"itemSize": 3,
				"type": "Float32Array",
				"array": []
			},
			"normal": {
				"itemSize": 3,
				"type": "Float32Array",
				"array": []
			}
		}
	}
}
```

Thanks a lot for the help.

tpaviot, 2017-05-07:
@DEKHTIARJonathan The Tesselator class in pythonocc-0.17.3 works as expected according to the tests executed on all platforms/architecture. It may segfault according to the shape passed as an argument. I guess the shape on the screenshot comes from a STEP file, can you please attach this shape so that it can be tested.

The "export to threejs javascript file" feature was introduced at the very early prototyping stage of the threejs support. It was actually a *very* ugly way to proceed, and not suitable for large models. The "import json" feature is much better:

* json parsers are very fast. A huge json file (dozains of megabytes) can be loaded within seconds ;

* json files can be loaded asynchronously ;

* it easy to split a big shapes into subshapes using json ;

* it's safer to load a json rather than importing/executing js code.

As a consequence, I strongly encourage you port your code to the new json importer. This will result in a better program.

DEKHTIARJonathan, 2017-05-11:
Hello @tpaviot ,

You will find on that link 11 STEP files in the zip files available at this address: [](https://www.amazon.de/clouddrive/share/xCbZLGnnmEad1o0WK8ufrWB7GwAIyooieJf9Eq3WJNs?ref_=cd_ph_share_link_copy)

I have a very weird behavior. I try to loop over a list of files. Load them one by one, take a screenshot and export them as ThreeJS (+X3Dom if possible).

However the program crash randomly during the export, more often with some specific files (Actuator002.step for instance). And even when it works, JSON String + X3Dom Files are almost almost empty as followed :  

JSON String : 
```
{
        "metadata": {
                "version": 4.4,
                "type": "BufferGeometry",
                "generator": "pythonOCC"
        },
        "uuid": "data1/Actuator/Actuator006.step",
        "type": "BufferGeometry",
        "data": {
        "attributes": {
                        "position": {
                                "itemSize": 3,
                                "type": "Float32Array",
                                "array": []
                        },
                        "normal": {
                                "itemSize": 3,
                                "type": "Float32Array",
                                "array": []
                        }
                }
        }
}
```

XML File:

```
<?xml version='1.0' encoding='UTF-8'?>
<!DOCTYPE X3D PUBLIC 'ISO//Web3D//DTD X3D 3.1//EN' 'http://www.web3d.org/specifications/x3d-3.1.dtd'>
<X3D>
	<Head>
		<meta name='generator' content='pythonOCC, http://www.pythonocc.org'/>
	</Head>
	<Scene>
		<Shape>
			<Appearance>
				<Material DEF='Shape_Mat' diffuseColor='0.65 0.65 0.65' shininess='0.9' specularColor='1 1 1'></Material>
			</Appearance>
			<TriangleSet solid='false'>
				<Coordinate point=''></Coordinate>
				<Normal vector=''></Normal>
			</TriangleSet>
		</Shape>
	</Scene>
</X3D>
```

Thanks a lot for the help. I can send you the program if you want too. It's a short script to batch conversion from STEP to ThreeJS and X3Dom.

DEKHTIARJonathan, 2017-05-15:
@tpaviot any update on that topic ?

DEKHTIARJonathan, 2017-05-15:
Hello @tpaviot, I have created for you a reproducible example of the issue:

The archive file can be downloaded here: https://www.amazon.de/clouddrive/share/VBTcuR3e31aekL03V4FAwO8zszlfDn3p49mW9BCCmbu?ref_=cd_ph_share_link_copy

The archive contains : 

- **reproducible_crash.py**: A script based on http://www.pythonocc.org/quick-examples/step-ap203-import/ reproducing the error.
- 4 STEP CAD Models provided by the NIST as example for the different "standards" of the STEP Norm. These models are perfectly clean and working.
   - AP203
   - AP203e2
   - AP214
   - AP242
- 1 custom STEP model that comes from my library, which also crash.

I hope it will be sufficient for you to handle the situation.

I use : 
- Windows 10 - 64bits
- Python 3.6.1 :: Anaconda (64-bit)
- pythonocc-core  0.17.3  py36_0    pythonocc
- oce 0.17.3  vc14_1  [vc14]  oce
- freeimageplus  3.17.0  vc14_0  [vc14]  dlr-sc
- freetype  2.6.3 vc14_1  [vc14]  dlr-sc
- gl2ps 1.3.8  vc14_3  [vc14]  dlr-sc
- tbb 4.3.6 vc14_1  [vc14]  dlr-sc

tpaviot, 2017-05-16:
@DEKHTIARJonathan ok I'll process as soon as possible

tpaviot, 2017-05-16:
@DEKHTIARJonathan The tesselation must explicitely be computed, i.e. explicitely call the .Compute() method other wise the vertex/face lists are empty, thus causing empty files or crashes. The right way to proceed is:

````
tess = Tesselator(aResShape)
tess.Compute()
if run_threejs_export:
    threejs_string = tess.ExportShapeToThreejsJSONString(model_list[index])
    print(threejs_string)
        with open("outputs/threejs/shape_"+model_list[index]+".json", "w") as f: 
            f.write(threejs_string)
````
Compute method takes two optional parameters: compute_edges, a boolean, and mesh_quality a float.

DEKHTIARJonathan, 2017-05-16:
Thanks a lot for your help. 
And it is perfectly working with Python 2.7.13 and Python 3.6.1

Is this documented somewhere that it is mandatory to apply Tessalator(shape).Computer() with the different options available and their impacts ?

tpaviot, 2017-05-17:
@DEKHTIARJonathan It's not documented in a docstring. You have an example about mesh_quality and contour edges display at https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_webgl_mesh_quality.py

The right place for a docstring should be in the SWIG file, here https://github.com/tpaviot/pythonocc-core/blob/master/src/Visualization/Visualization.i#L65 Feel free to submit a patch with related docstring

closed 2017-05-17 by tpaviot