tpaviot/pythonocc-core #422: Voxel example does'nt display anything

tpaviot, 2017-05-06:
On windows, py27/32, the following doesn't display anything whereas I would expect a nice voxel to be displayed in the graphic Window. What's wrong ?

````
#!/usr/bin/env python

##Copyright 2017 Thomas Paviot (tpaviot@gmail.com)
##
##This file is part of pythonOCC.
##
##pythonOCC is free software: you can redistribute it and/or modify
##it under the terms of the GNU Lesser General Public License as published by
##the Free Software Foundation, either version 3 of the License, or
##(at your option) any later version.
##
##pythonOCC is distributed in the hope that it will be useful,
##but WITHOUT ANY WARRANTY; without even the implied warranty of
##MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
##GNU Lesser General Public License for more details.
##
##You should have received a copy of the GNU Lesser General Public License
##along with pythonOCC.  If not, see <http://www.gnu.org/licenses/>.

from OCC.BRepPrimAPI import BRepPrimAPI_MakeSphere
from OCC.Voxel import (Voxel_BoolDS, Voxel_FastConverter, Voxel_Prs,
                      Voxel_ColorDS, Voxel_VDM_POINTS)
from OCC.Quantity import Quantity_Color, Quantity_NOC_AZURE

from OCC.Display.SimpleGui import init_display

display, start_display, add_menu, add_function_to_menu = init_display()

sphere_shp = BRepPrimAPI_MakeSphere(12., 12., 12.).Shape()
 
print("Perform shape voxelization ...")
ds = Voxel_ColorDS()
converter = Voxel_FastConverter(sphere_shp, ds, 0.1, 10, 10, 10, 1)
ret = converter.Convert()
print("done.")

# display the Voxel
vox_prs = Voxel_Prs()
vox_prs.SetColorVoxels(ds)
vox_prs.SetDisplayMode(Voxel_VDM_POINTS)
vox_prs.SetColor(Quantity_Color(Quantity_NOC_AZURE))
vox_prs.SetPointSize(1.)
vox_prs.SetQuadrangleSize(100)
vox_prs.SetTransparency(0.5)
display.Context.Display(vox_prs.GetHandle(), True)
# let's display the voxel
display.FitAll()
start_display()
````

DEKHTIARJonathan, 2017-05-15:
it seems like Python 3.6 version is broken too:

```
Traceback (most recent call last):
  File "test.py", line 21, in <module>
    from OCC.Voxel import Voxel_BoolDS, Voxel_FastConverter, Voxel_Prs, Voxel_ColorDS, Voxel_VDM_POINTS
ModuleNotFoundError: No module named 'OCC.Voxel'
```

shivangag, 2018-05-31:
@tpaviot Any update on this issue? I am also not able to visualize voxels.

jf---, 2018-05-31:
IIRC the `Voxel` is no longer supported in the underlying C++ lib ( OCCT 6.9.1 / OCE 0.18 )

shivangag, 2018-05-31:
Sad to hear that.
I was trying to voxelize a .stp file, exported from CATIA, to convert it into a 3 dimensional numpy array, which can then be used as an input to a neural network with 3D convolutional layers.
Can you suggest any other method to achieve this conversion?
Thanks.

tpaviot, 2020-01-05:
Deprecated. Voxel package is not available anymore in occt.

closed 2020-01-05 by tpaviot