tpaviot/pythonocc-core #424: conda fail on OSX64

IronwoodLabs, 2017-05-18:
I created a clean Python 3 conda environment and ran the following.

ironMac:~ myusername$ source activate pythonocc

(pythonocc) ironMac:~ myusername$ conda install -c oce -c pythonocc pythonocc-core=0.17.3
Fetching package metadata ...............
Solving package specifications: 

PackageNotFoundError: Dependencies missing in current osx-64 channels: 
  - pythonocc-core 0.17.3* -> oce ==0.17.3 -> freeimageplus
  - pythonocc-core 0.17.3* -> oce ==0.17.3 -> gl2ps
  - pythonocc-core 0.17.3* -> oce ==0.17.3 -> tbb
  - pythonocc-core 0.17.3* -> pyobjc-core

Close matches found; did you mean one of these?

    freeimageplus: freeimage
    pyobjc-core: pythonocc-core
    tbb: tblib

(and similarly for the other packages)

How can I manually install dependencies for OSX64?

DEKHTIARJonathan, 2017-05-18:
Hello @IronwoodLabs ,

Can you try running the following:
```
conda install -y -c dlr-sc freeimageplus=3.17.0 gl2ps=1.3.8 tbb=4.3.6
conda install -y -c conda-forge freetype=2.7
conda install -y -c pythonocc -c oce pythonocc-core=0.17.3 python=3
```

tpaviot, 2017-05-21:
@IronwoodLabs Can u please report success/failure

adam-urbanczyk, 2017-05-28:
Same issue on linux64:

```
conda create -n pythonocc-test -c pythonocc -c oce pythonocc-core==0.17.3 python=2
Fetching package metadata .............
Solving package specifications: 

PackageNotFoundError: Dependencies missing in current linux-64 channels: 
  - pythonocc-core ==0.17.3 -> oce ==0.17.3 -> freeimageplus
  - pythonocc-core ==0.17.3 -> oce ==0.17.3 -> gl2ps
  - pythonocc-core ==0.17.3 -> oce ==0.17.3 -> tbb

Close matches found; did you mean one of these?

    freeimageplus: freeimage
    tbb: tblib

(and similarly for the other packages)
```

The workaround for me was:

```
conda create -n pythonocc-test -c conda-forge -dlr-sc -c pythonocc -c oce pythonocc-core==0.17.3 python=2
```

which results in the following env:

```
ca-certificates           2017.4.17                     0    conda-forge
certifi                   2017.4.17                py27_0    conda-forge
dbus                      1.10.10                       3    conda-forge
expat                     2.1.0                         2    conda-forge
fontconfig                2.12.1                        4    conda-forge
freeimageplus             3.17.0                        0    dlr-sc
freetype                  2.7                           1    conda-forge
gettext                   0.19.7                        1    conda-forge
gl2ps                     1.3.8                         1    dlr-sc
glib                      2.51.4                        0    conda-forge
gst-plugins-base          1.8.0                         0    conda-forge
gstreamer                 1.8.0                         1    conda-forge
icu                       58.1                          1    conda-forge
jpeg                      9b                            0    conda-forge
libffi                    3.2.1                         3    conda-forge
libiconv                  1.14                          4    conda-forge
libpng                    1.6.28                        0    conda-forge
libxcb                    1.12                          1    conda-forge
libxml2                   2.9.4                         4    conda-forge
ncurses                   5.9                          10    conda-forge
oce                       0.17.3                        1    oce
openssl                   1.0.2k                        0    conda-forge
pcre                      8.39                          0    conda-forge
pip                       9.0.1                    py27_0    conda-forge
pyqt                      5.6.0                    py27_3    conda-forge
python                    2.7.13                        1    conda-forge
pythonocc-core            0.17.3                   py27_0    pythonocc
qt                        5.6.2                         2    conda-forge
readline                  6.2                           0    conda-forge
setuptools                33.1.1                   py27_0    conda-forge
sip                       4.18                     py27_1    conda-forge
sqlite                    3.13.0                        1    conda-forge
tbb                       2017_20170226                 0    conda-forge
tk                        8.5.19                        1    conda-forge
wheel                     0.29.0                   py27_0    conda-forge
xorg-libxau               1.0.8                         3    conda-forge
xorg-libxdmcp             1.1.2                         3    conda-forge
xz                        5.2.2                         0    conda-forge
zlib                      1.2.11                        0    conda-forge
```

closed 2017-06-06 by tpaviot