tpaviot/pythonocc-core #430: Install OCE/PYTHONOCC on Macosx

fperignon, 2017-05-30:
Macosx 10.11.6

Install from source of oce 0.17.3 : ok
but pythonocc 0.17.3 fails:

Undefined symbols for architecture x86_64:  "TDataXtd_Axis::Set(TDF_Label const&)", referenced from:      _wrap_TDataXtd_Axis_Set__SWIG_0(_object*, int, _object**) in TDataXtdPYTHON_wrap.cxx.o

fperignon, 2017-05-30:
Solution : use conda.

- install anaconda using graphic installer (see https://docs.continuum.io/anaconda/install-macos.html)
- install python occ : 
conda create -n pythonocc -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.17.3 python=3

- compile siconos (cxx-11 mode):
cmake ... -DCMAKE_CXX_STANDARD=11 -DSICONOS_USE_BOOST_FOR_CXX11=OFF 

Error :

Undefined symbols for architecture x86_64:
  "math_Vector::Dump(std::__1::basic_ostream<char, std::__1::char_traits<char> >&) const", referenced from:
      mymath_FunctionSetRoot::Dump(std::__1::basic_ostream<char, std::__1::char_traits<char> >&) const in mymath_FunctionSetRoot.cxx.o
ld: symbol(s) not found for architecture x86_64

Problem comes from Dump function in 
/mechanics/src/mechanisms/CADMBTB/mymath_FunctionSetRoot.cxx 
used to overload ostream op.

Found something here:
https://www.opencascade.com/content/problems-getting-output-file-dump-method

Any idea of a proper solution?

tpaviot, 2017-06-14:
@fperignon I have no idea about what's wrong. Is Osx 10.11.6 the latest available version ? I used to work on a MacbookPro some years ago, but I don't have any osx machine anymore. The conda binaries are compiled using the travis-ci osx virtual machine, I can't drive any additionnal test.

Can you please run the unittests that come with pythonocc ?

rainman110, 2017-06-29:
@fperignon @tpaviot The pythonocc packages are compiled with MACOSX_DEPLOYMENT_TARGET=10.8.

This means, that no C++11 is available and the libstdc++ is beeing used by OCE. When you compile siconos, it will probably use libc++ with C++11 support. Hence, you'll get linker errors.

When you want to link against the conda packages, you have to use MACOSX_DEPLOYMENT_TARGET=10.8 and disable C++11.

tpaviot, 2017-09-14:
The latest master branch (future pythonocc-0.18.1) uses MACOSX_DEPLOYMENT_TARGET=10.9, and is linked against libc++

closed 2017-09-14 by tpaviot