tpaviot/pythonocc-core #431: Problems using pythonocc in Jupyter

boustrephon, 2017-06-01:
I am running conda 4.3.18 in an Anaconda3 64-bit installation on Win7 at work. I have some admin permissions.

I followed the instruction to create a new pythonocc environment (great idea to provide this...):
`conda create -n pythonocc -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.17.3 python=3`

I note that swig is not in the installation.

I installed ipykernel and created an ipykernel for use in jupyter:
`python -m ipykernel install --user --name pythonocc --display-name "Python (pyOCC)"`

I decided to try out the Simple Box example. I was able to import the init_display, but when I tried to import the BRepPrimAPI_MakeBox, I got the following error:

```
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-3-a9f866066e05> in <module>()
----> 1 from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox
      2 #display, start_display, add_menu, add_function_to_menu = init_display()

C:\ProgramData\Anaconda3\envs\pythonocc\lib\site-packages\OCC\BRepPrimAPI.py in <module>()
     20         mname = '.'.join((pkg, '_BRepPrimAPI')).lstrip('.')
     21         return importlib.import_module(mname)
---> 22     _BRepPrimAPI = swig_import_helper()
     23     del swig_import_helper
     24 elif version_info >= (2, 6, 0):

C:\ProgramData\Anaconda3\envs\pythonocc\lib\site-packages\OCC\BRepPrimAPI.py in swig_import_helper()
     19         pkg = __name__.rpartition('.')[0]
     20         mname = '.'.join((pkg, '_BRepPrimAPI')).lstrip('.')
---> 21         return importlib.import_module(mname)
     22     _BRepPrimAPI = swig_import_helper()
     23     del swig_import_helper

C:\ProgramData\Anaconda3\envs\pythonocc\lib\importlib\__init__.py in import_module(name, package)
    124                 break
    125             level += 1
--> 126     return _bootstrap._gcd_import(name[level:], package, level)
    127 
    128 

ImportError: DLL load failed: The specified module could not be found.
```

Did I miss something? How can I fix this? I tried installing swig to see if that would fix it, but it didn't make any difference. Any help would be appreciated. Thanks.

boustrephon, 2017-06-01:
Continuing with the [dumb box example](http://www.pythonocc.org/quick-examples/hello-dumb-box/):
I also checked to see if the display function would work:

As I mentioned, the import did not complain:
`from OCC.Display.SimpleGui import init_display`

However, when I tried to define the display...
`display, start_display, add_menu, add_function_to_menu = init_display()`

... I got the following errors:
```
INFO:OCC.Display.backend:backend loaded: qt-pyqt5
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt5
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-4-7053f85cee0b> in <module>()
----> 1 display, start_display, add_menu, add_function_to_menu = init_display()

C:\ProgramData\Anaconda3\envs\pythonocc\lib\site-packages\OCC\Display\SimpleGui.py in init_display(backend_str, size)
    143     # Qt based simple GUI
    144     elif 'qt' in used_backend:
--> 145         from OCC.Display.qtDisplay import qtViewer3d
    146         QtCore, QtGui, QtWidgets, QtOpenGL = get_qt_modules()
    147 

C:\ProgramData\Anaconda3\envs\pythonocc\lib\site-packages\OCC\Display\qtDisplay.py in <module>()
     27 log = logging.getLogger(__name__)
     28 
---> 29 from OCC.Display import OCCViewer
     30 from OCC.Display.backend import get_qt_modules
     31 

C:\ProgramData\Anaconda3\envs\pythonocc\lib\site-packages\OCC\Display\OCCViewer.py in <module>()
     25 import itertools
     26 
---> 27 from OCC.AIS import AIS_MultipleConnectedInteractive, AIS_Shape
     28 from OCC.TopoDS import TopoDS_Shape
     29 from OCC.gp import gp_Dir, gp_Pnt, gp_Pnt2d, gp_Vec

C:\ProgramData\Anaconda3\envs\pythonocc\lib\site-packages\OCC\AIS.py in <module>()
     20         mname = '.'.join((pkg, '_AIS')).lstrip('.')
     21         return importlib.import_module(mname)
---> 22     _AIS = swig_import_helper()
     23     del swig_import_helper
     24 elif version_info >= (2, 6, 0):

C:\ProgramData\Anaconda3\envs\pythonocc\lib\site-packages\OCC\AIS.py in swig_import_helper()
     19         pkg = __name__.rpartition('.')[0]
     20         mname = '.'.join((pkg, '_AIS')).lstrip('.')
---> 21         return importlib.import_module(mname)
     22     _AIS = swig_import_helper()
     23     del swig_import_helper

C:\ProgramData\Anaconda3\envs\pythonocc\lib\importlib\__init__.py in import_module(name, package)
    124                 break
    125             level += 1
--> 126     return _bootstrap._gcd_import(name[level:], package, level)
    127 
    128 

ImportError: DLL load failed: The specified module could not be found.

```

adam-urbanczyk, 2017-06-01:
An idea: check what is in ```mname``` variable  (I assume some *pyd file) and then inspect it with dependancywalker.

tpaviot, 2017-06-02:
@boustrephon Perhaps a conflict with your freetype version. Try to upgrade to 2.7/conda-forge:
````
$ conda install -c conda-forge freetype=2.7
````

boustrephon, 2017-06-02:
Hi Thomas, thanks for the quick reply. I have checked, and freetype is
already at v2.7.

boustrephon, 2017-06-02:
Hi Adam,

Thanks for the reply. I am not so experienced with this kind of thing, so I
hope I have followed the correct steps.

   1. From what I can tell, mname in this case is referring to
   OCC._BRepPrimAPI.
   2. When I look at the file in Dependency Walker I get a series of
   warnings about the system not finding the files specified (see attached
   screenshot).

I presume this means that there are some missing dependencies. Would I be
right in assuming that I need to install TKinter? If so, this would
presumably deal with half of the errors.

Any guidance you can give would be much appreciated.

Andrew

adam-urbanczyk, 2017-06-05:
@boustrephon  e.g. tkbrep.dll is oce related (not tkinter). I can reproduce your problem on Windows 10 with or without ipython. If I have some time, I'll try to look at this.

boustrephon, 2017-06-06:
Thank you very much.

adam-urbanczyk, 2017-06-07:
@boustrephon  I think I found the problem - wrong version of tbb was installed. The workaround for me was to install correct version of tbb package (the dll load error was due to wrong version of tbb installed):

```conda install -c dlr-sc tbb=4.3.6```

Can you confirm that this works for you?

@tpaviot  I am no conda expert but I think the oce package should specify a version of tbb in the run requirements. Would you agree?

boustrephon, 2017-06-08:
Hi Adam, thank you. Yes, that has fixed it like magic. Thank you very much. I can now get on to seeing how I can use this with ifcopenshell.

tpaviot, 2017-06-10:
@adam-urbanczyk I agree, oce-0.17.3 conda distro didn't explicetly set dependencies versions. It's different for oce-0.18.1 (you can see the conda recipes for both versions https://github.com/tpaviot/pythonocc-core/blob/build-oce/oce-0.17.3/meta.yaml and https://github.com/tpaviot/pythonocc-core/blob/build-oce/oce-0.18.1/meta.yaml)

closed 2017-06-10 by tpaviot