tpaviot/pythonocc-core #435: How to display shape names or labels from STEP file?

sebastienaubry, 2017-06-21:
Hi,
I have imported a STEP file which contains labels:
...
#174=PRODUCT('NAME1','NAME1','',(#2)) ;
#1323=PRODUCT('NAME2','NAME2','',(#2)) ;
#1402=PRODUCT('NAME3','NAME3','',(#2)) ;
...

Using the dumpTopology example, I could display the tree of shapes contained in my file. But I would also like to display these labels, in order to select the shapes inside the same "compound". Is there a simple attribute or method on the shape object in order to get its label?
Thanks

hinxx, 2017-06-21:
Have a look at the https://github.com/tpaviot/pythonocc-core/pull/434. There is a method that can get the label name from STEP. Displaying that name in viewer is another story.

sebastienaubry, 2017-06-22:
Thank you. Using this example, I was able to retrieve the all labels in my STEP file.

Now I would like to export an assembly with a given label as a list of (x,y) polygons vertices. Is there an easy way to do that?

tpaviot, 2017-06-23:
@sebastienaubry can you please open up a new issue or ask the question on the pythonocc google group

closed 2017-06-23 by tpaviot