tpaviot/pythonocc-core #436: Installation of `pythonocc-core` with `conda`

yw5aj, 2017-06-21:
First of all - thank you for the great package!

I was playing with it and realized that to install `pythonocc-core` with `conda`, I need to do the following:

```
conda install -c conda-forge -c dlr-sc -c oce -c pythonocc pythonocc-core
```

Where the:
`-c pythonocc pythonocc-core` is obvious;
`-c oce` is for `oce` and `smesh`;
`-c dlr-sc` is for `freeimageplus`, `gl2ps` and `tbb`;
`-c conda-forge` is for `rtree`, `scikit-image` and `vtk`.

Hope that it helps for someone else who will see this!

Shawn

tpaviot, 2017-06-23:
Thank you @yw5aj , but this is already available in the README.md file.

closed 2017-06-23 by tpaviot

yw5aj, 2017-06-23:
Oh yes! You are right. Sorry about that - I was on [http://www.pythonocc.org/download/](http://www.pythonocc.org/download/) and that page was outdated. The `README.md` file is exactly what I was looking for!