tpaviot/pythonocc-core #437: Import error on Windows 7

trelau, 2017-06-27:
I have a fork of pythonocc-core where I integrated SMESH 7.7.1 and Netgen 6.2 and put some Python 3.5 Windows 64-bit packages up on Anaconda. This worked great until I updated the pythonocc-core master and built Netgen and SMESH with OCE 0.18.1 (I used the old SMESH wrappers, not the 6.7.X ones currently in pythonocc). Everything is built on appveyor.

On my Windows 10 64-bit machine with Python 3.5, things still operate like normal, but there have been reports of users with Windows 7 getting the following error:

ImportError: DLL load failed: The specified module could not be found.
return _bootstrap._gcd_import(name[level:], package, level)
line 26 in import_module
line 21 in swig_import_helper
_SMESH = swig_import_helper()

It seems importing SMESH (and its other binaries like SMDS) is throwing this error and it seems to be isolated to Windows 7.

Since this seems to be an issue on a forked effort, I'm just hoping to get a few ideas of the problem based on anyone's experience. It's been quite a battle trying to figure it out. Thanks for the help.

rainman110, 2017-07-04:
@trelau Try to inspect the SMESH module for missing or incompatible DLLs with the [Dependency Walker](http://www.dependencywalker.com/).

This should give a hint, what the actual problem is.

trelau, 2017-07-06:
@rainman110 thank you. turns out the version of VTK I am building with is trying to link to debug binaries of ucrtbase, vcruntime, and one other. Once we copied those into the OCC folder it worked. For some reason, in previous versions I was building with VTK 7.0.0 from dlr-sc channel, but now it builds with VTK 7.0.0 from conda-forge channel and I can't figure out how to force it to use dlr-sc again. Maybe the conda-forge build used the debug libraries? Is there a way to force a specific channel on appveyor or even in conda in general? I added dlr-sc channel last in appveyor which i think gives it highest priority, but it still tries to use conda-forge VTK 7.0.0.

rainman110, 2017-07-07:
@trelau From conda 4.1.0, you can define channel priorities.

Therefore, you should set the proirity of the dlr-sc channel  higher than the conda-forge channel.

Since you already added dlr-sc last, it should have highest priority. Maybe the build version of VTK from conda forge is higher than from my dlr-sc channel?

Nevertheless, I think it is a bug, if the VTK libs are compiled in debug mode. It's probably best if you contact the owner of the VTK package.

trelau, 2017-07-07:
@rainman110 thanks. When I build smesh it always uses "vtk-7.0.0-py35_vc14_1" from conda-forge channel even though there is a "vtk-7.0.0-py35_vc14_2." available from both conda-forge and dlr-sc. I don't know why it keep using the "_1" version. It must be one of the other requirements I'm specifying.

closed 2019-04-28 by trelau