tpaviot/pythonocc-core #438: get_backend function not found

kozintsev, 2017-06-27:
Hello!
I watched python-occ examples and in...
_pythonocc-core/examples/core_visualization_overpaint_viewer.py line 41-43_
get_backend function not found

Yours faithfully!

chavan-vjti, 2017-07-28:
Same error on my side 👎 
any solution to his issue?

kozintsev, 2017-07-28:
@chavan-vjti look here:
https://github.com/kozintsev/ResearchPythonOCC/blob/master/InteractiveEditor/TestQtBackendNative.py

Maybe it will help you.

chavan-vjti, 2017-07-28:
@kozintsev 
Thank you for your help.
I'll let you about this.
Thank you!

chavan-vjti, 2017-07-28:
@kozintsev 
I tried the above mentioned python program that you suggested but nothing gets displayed in the Python OCC screen. It remains blank even though we've mentioned the command as, "print('hello')".
Any reason ?

tpaviot, 2017-08-07:
@kozintsev fixed by PR #447 Thanks for your report.

closed 2017-08-07 by tpaviot