tpaviot/pythonocc-core #439: Menus do not work (well) in SimpleGui under macOs

krontzo, 2017-06-28:
I've created two environments in conda:

1) pythonocc-core            0.17.3                   py36_0    pythonocc
2) pythonocc-core            0.18                     py27_0    pythonocc

If you add menu entries like in:

```python
if __name__ == '__main__':
    display, start_display, add_menu, add_function_to_menu = init_display()
    add_menu('STEP import')
    add_function_to_menu('STEP import', run)
    add_function_to_menu('STEP import', exit)
    start_display()
```

menus do not activate, unless you switch to other application and switch back to SimpleGui Window; then menus do work.

Tested with conda Python3.6/PythonOCC0.17.3  and  Python2.7/PythonOCC0.18  under macOs Sierra 10.12.5.

Also, tetrahedron labels (X, Y, Z) do not show up in any environment.

rainman110, 2017-06-29:
@krontzo I also have the problems with the trihedron labels - but only on my hackintosh. On my "Genuine" mac, this is not a problem. Do you also use a hackintosh?

rainman110, 2017-07-02:
@krontzo I also can confirm the problems with the qt menus.

krontzo, 2017-07-03:
@rainman110 I use a mba and a mini. Same problem

rainman110, 2017-07-06:
Okay, the labels might be missing because OCE does not find the font files on OS X. This is decribed in the following OCCT issue:
http://git.dev.opencascade.org/gitweb/?p=occt.git;a=commitdiff;h=264abd72f2508894ff6d514f2d9ff5c2443656f8;hp=e5260e1dfa44d1ca7adcdc0b04b3dd7506ffa103

@tpaviot  We should backport this patch tp the current OCE version.

rainman110, 2017-07-06:
Okay, the labels might be missing because OCE does not find the font files on OS X. This is decribed in the following OCCT issue:

https://tracker.dev.opencascade.org/view.php?id=27505

This is the corresponding patch for the issue.

http://git.dev.opencascade.org/gitweb/?p=occt.git;a=commitdiff;h=264abd72f2508894ff6d514f2d9ff5c2443656f8;hp=e5260e1dfa44d1ca7adcdc0b04b3dd7506ffa103

@tpaviot  We should backport this patch tp the current OCE version.

This also explains, why on my MBP the fonts are working and on my hackintosh not. On the MBP, I installed X-Quartz ( the mac X11 implementation). This provides the required font paths. If X11 is not installed, the fonts are missing.

tpaviot, 2017-07-09:
@rainman110 sure, submit a patch at the oce github repos, we'll merge it into master for a 0.18.2 release.

tpaviot, 2017-09-11:
@rainman110 Can you please try the latest nigthly build (conda channel tpaviot), it's built upon oce-0.18.2 including the requested change.

tpaviot, 2017-12-01:
No feedback, I assume it(s fixed. Feel free  @krontzo and @rainman110 to reopen this issue if needed

closed 2017-12-01 by tpaviot