tpaviot/pythonocc-core #440: white screen appears during animation of some complicated shapes.

huntsmanydw, 2017-06-29:
problem:white screen appears  during animation of some complicated shapes.

The pythonocc version I used is "pythonocc-core-0.17.2"

I followed the code in the file of "core_animation.py" to do some motion simulation project. I faced a white-sceen problem during the motion process.

The code "core_animation.py" works well, but if I load some complicate *.step format files to do the simulation. During the motion process, the display windows becomes white screen. 

I tried the following ways to solve this problem, but failed.
try 1: add some delay during the animation loop, failed.

try 2: setDoublebuffer ON, failed.

I searched the document, 
in the "OCC.Display.OCCViewer module", this is a function named by SetDoubleBuffer(). from the comments, it seems that if I can setDoublebuffer ON, this problem can be solved.

in file "\pythonocc-core-0.17.2\src\Display\OCCViewer.py"
    def SetDoubleBuffer(self, on_or_off):
        """enables double buffering when shapes are moved in the viewer
        a very shaky picture is drawn, since double buffering
        is disabled by default. fixed here..."""
        self.View.Window().GetObject().SetDoubleBuffer(bool(on_or_off))

Unfortunitely, I failed to setDoublebuffer ON.

I added this line 

display, start_display, add_menu, add_function_to_menu = init_display("qt-pyqt5")
display.SetDoubleBuffer(True) #add this line

the following errors appears:
Traceback (most recent call last):
  File "VRs_V2.2.5.py", line 84, in <module>
    display.SetDoubleBuffer(True)
  File "C:\Users\dwyan\Anaconda2\lib\site-packages\OCC\Display\OCCViewer.py", line 173, in SetDoubleBuffer
    self.View.Window().GetObject().SetDoubleBuffer(bool(on_or_off))
AttributeError: 'Aspect_Window' object has no attribute 'SetDoubleBuffer'

 ----------------------------------------------------------

#my code structure is like this: 

display, start_display, add_menu, add_function_to_menu = init_display("qt-pyqt5")

step_shp = read_step_file('somestepfile.STEP')
ais_boxshp = display.DisplayShape(step_shp, update=True)

for I in range(N):

	aTrsf = gp_Trsf()
	
	#change the value of v1, v2, v3, …, v12
	aTrsf.SetValues(v1, v2, v3, …, v12)
	aToploc = TopLoc_Location(aTrsf)    
	display.Context.SetLocation(ais_boxshp, aToploc) 
	display.Context.UpdateCurrentViewer()   
	
	
	

Hope someone can help me! Thanks a lot.

tpaviot, 2017-09-19:
SetDoubleBuffer is an outdated method (I removed it from the current master branch), the oce opengl graphic driver automatically switches double buffering on if available. How do you define "some complicated shape "?

tpaviot, 2017-09-19:
@huntsmanydw Tested with the "cylinder head" example, I did not notice any problem. Note : if you want to get a smoother animation, switch off antialiasing by hitting the "b" before launching the animation

closed 2017-09-19 by tpaviot