tpaviot/pythonocc-core #441: Import STEP with name

tnakaicode, 2017-07-03:
I want to get each walls' name (Wall-1, Wall-2, Wall-3, Wall-4). Please look at .png file (CATIA GUI capture).
This step file is surfaces of duct. These surface are flat. I want to get both each surface's vertex and it' s name.

But, I don't know how to get each surface's name.
I want your opinions.

[image]

[BSM-5.zip](https://github.com/tpaviot/pythonocc-core/files/1118262/BSM-5.zip)

rainman110, 2017-07-05:
@tnakaicode May I redirect to our implementation of the step reading process:

https://github.com/DLR-SC/tigl/blob/master/src/imports/CTiglStepReader.cpp#L47

It currently reads only the shape name, but can be also changed to read face names and so on.

Our step writer writes also face names, therefore you might also look into it:
https://github.com/DLR-SC/tigl/blob/master/src/exports/CTiglExportStep.cpp#L204

And if you want to look into a full blown implementation, the StepCAFControlReader from OCE could become handy! 
https://github.com/tpaviot/oce/blob/master/src/STEPCAFControl/STEPCAFControl_Reader.cxx

tpaviot, 2017-07-09:
@tnakaicode have a look at the core_load_step_with_color.py example (https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_load_step_with_colors.py), based upon the STEPCAFControl_Reader.

tnakaicode, 2017-07-10:
This is output of core_load_step_with_color.py example  code (https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_load_step_with_colors.py).
But, it cannot identify 4 walls, only root product (Part4).

-----------------------------------------------------
INFO:OCC.Display.backend:backend loaded: qt-pyqt5
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt5
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
WNT window created.
Display3d class successfully initialized.
 ########################################

Number of shapes at root : 1

[2] level 0, handling LABEL Part4

0:1:1:1 NOT imported; NOT modified; NO attribute modified; has 3 attributes.
        TNaming_NamedShape      Trans. 0; Valid;        ID = c4ef4200-568f-11d1-8940-080009dc3333
        XCAFDoc_ShapeMapTool    Trans. 0; Valid;        ID = 3b913f4d-4a82-44ef-a0bf-9e01e9ff317a
        TDataStd_Name   Trans. 0; Valid;        ID = 2a96b608-ec8b-11d0-bee7-080009dc3333

Is Assembly    : False
Is Free        : True
Is Shape       : True
Is Compound    : False
Is Component   : False
Is SimpleShape : True
Is Reference   : False
Nr Users       : 0
Nb subshapes   : 0
Nb components  : 0

########  simpleshape label : class<'TDF_Label'>
    all ass locs   : []
    FINAL loc    :
    tran form    : 0
    rotation     : class<'gp_Quaternion'>
    X            : 0.0
    Y            : 0.0
    Z            : 0.0
    W            : 1.0
    translation  : class<'gp_XYZ'>
    X            : 0.0
    Y            : 0.0
    Z            : 0.0

Handled 2 labels

-----------------------------------------------------

tnakaicode, 2017-07-12:
This is output of core_load_step_with_color.py example code (https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_load_step_with_colors.py).
But, it cannot identify 4 walls, only root product (Part4).

---------------------------------------

INFO:OCC.Display.backend:backend loaded: qt-pyqt5
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt5

3D rendering pipe initialisation

Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
WNT window created.
Display3d class successfully initialized.
########################################

Number of shapes at root : 1

[2] level 0, handling LABEL Part4

0:1:1:1 NOT imported; NOT modified; NO attribute modified; has 3 attributes.
TNaming_NamedShape Trans. 0; Valid; ID = c4ef4200-568f-11d1-8940-080009dc3333
XCAFDoc_ShapeMapTool Trans. 0; Valid; ID = 3b913f4d-4a82-44ef-a0bf-9e01e9ff317a
TDataStd_Name Trans. 0; Valid; ID = 2a96b608-ec8b-11d0-bee7-080009dc3333

Is Assembly : False
Is Free : True
Is Shape : True
Is Compound : False
Is Component : False
Is SimpleShape : True
Is Reference : False
Nr Users : 0
Nb subshapes : 0
Nb components : 0

######## simpleshape label : class<'TDF_Label'>
all ass locs : []
FINAL loc :
tran form : 0
rotation : class<'gp_Quaternion'>
X : 0.0
Y : 0.0
Z : 0.0
W : 1.0
translation : class<'gp_XYZ'>
X : 0.0
Y : 0.0
Z : 0.0

Handled 2 labels

---------------------------------------

hinxx, 2017-07-12:
@tnakaicode I hacked the code a bit and were able to get what I think is four faces (walls) you are looking for. Far from complete solution but might give you an idea.

First I imported dumpTopology:

```
from core_topology_traverse import Topo, dumpTopology

```
Then in the getShapes() I added:

```
    shape = shape_tool.GetShape(root)
    dumpTopology(shape)
```

The resulting console output is:

```
Compound.: 54408449
..Shell: 51503393
....Face: 51496113
......Wire: 51511953
........Edge: 51505713
..........<Vertex 51498961: 8332.10025868 203.030189508 1561.87745913>
..........<Vertex 51499201: 8404.04838527 233.04741967 953.808961275>
........Edge: 51515041
..........<Vertex 51506705: 8845.0 417.015324562 1471.43939633>
..........<Vertex 51498961: 8332.10025868 203.030189508 1561.87745913>
........Edge: 51505489
..........<Vertex 51031585: 8845.0 417.015324562 1031.56062814>
..........<Vertex 51506705: 8845.0 417.015324562 1471.43939633>
........Edge: 54414833
..........<Vertex 51031585: 8845.0 417.015324562 1031.56062814>
..........<Vertex 51499201: 8404.04838527 233.04741967 953.808961275>
..Shell: 51495937
....Face: 51496913
......Wire: 54422561
........Edge: 54418273
..........<Vertex 54421361: 8845.0 417.015324562 1471.43939633>
..........<Vertex 54419121: 8332.10025868 203.030189508 1561.87745913>
........Edge: 54418513
..........<Vertex 54419121: 8332.10025868 203.030189508 1561.87745913>
..........<Vertex 54420177: 8332.10025868 -78.6904959686 1561.87745913>
........Edge: 53940353
..........<Vertex 54420593: 8845.0 135.294639086 1471.43939633>
..........<Vertex 54420177: 8332.10025868 -78.6904959686 1561.87745913>
........Edge: 53940977
..........<Vertex 54420593: 8845.0 135.294639086 1471.43939633>
..........<Vertex 54421361: 8845.0 417.015324562 1471.43939633>
..Shell: 54428145
....Face: 53939809
......Wire: 53939697
........Edge: 54418033
..........<Vertex 53939905: 8404.04838527 -48.6732658062 953.808961275>
..........<Vertex 53937105: 8332.10025868 -78.6904959686 1561.87745913>
........Edge: 53935665
..........<Vertex 53940113: 8845.0 135.294639086 1031.56062814>
..........<Vertex 53939905: 8404.04838527 -48.6732658062 953.808961275>
........Edge: 53937553
..........<Vertex 53940113: 8845.0 135.294639086 1031.56062814>
..........<Vertex 53934305: 8845.0 135.294639086 1471.43939633>
........Edge: 53938177
..........<Vertex 53934305: 8845.0 135.294639086 1471.43939633>
..........<Vertex 53937105: 8332.10025868 -78.6904959686 1561.87745913>
..Shell: 53939425
....Face: 53932481
......Wire: 53932673
........Edge: 53953089
..........<Vertex 53932721: 8404.04838527 233.04741967 953.808961275>
..........<Vertex 51497425: 8404.04838527 -48.6732658062 953.808961275>
........Edge: 53953329
..........<Vertex 53943905: 8845.0 417.015324562 1031.56062814>
..........<Vertex 53932721: 8404.04838527 233.04741967 953.808961275>
........Edge: 53953569
..........<Vertex 53943761: 8845.0 135.294639086 1031.56062814>
..........<Vertex 53943905: 8845.0 417.015324562 1031.56062814>
........Edge: 53950897
..........<Vertex 53943761: 8845.0 135.294639086 1031.56062814>
..........<Vertex 51497425: 8404.04838527 -48.6732658062 953.808961275>
```

tnakaicode, 2017-07-14:
I used Topo.Vertices() in module OCCUtils.Topology and can get every vertex's coordinate value. But, in this way, I cannot know the name of face which the vertex is in.

I want to get both the face's name and vertex's coordinate value in same time.

tpaviot, 2017-07-14:
@tnakaicode the Topo class only deals with topology, it has nothing to do with naming. I absolutely understand what is you need, you expressed it clearly, but I think you have enough material to reach your target. Don't expect our community to do the job for you. Please consider studying what @rainman110 and I suggested.

trelau, 2018-05-20:
@tnakaicode see closed issue https://github.com/tpaviot/pythonocc-core/issues/522. I think with that you can do what you need and close this issue.

jonititan, 2018-06-04:
https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_load_step_with_colors.py

closed 2019-02-19 by tnakaicode